whitequark/solvespace #4: Lathe group edges are not selectable

whitequark, 2015-03-06:
If you extrude a group, you can select a face and create a workplane on it. If you however lathe a group (sample data: http://rghost.net/75qGtXsJF), you cannot select any edges or faces. At the very least, edges and flat faces should be selectable.

whitequark, 2015-03-17:
Why does this happen? I'm not sure. It looks like SBezier::IsCircle doesn't return true for lathe'd edges.

Evil-Spirit, 2015-07-08:
refresh sample please

whitequark, 2015-07-08:
E.g. this: http://rghost.net/6rJsQrQns

The reason is just that the 3D entities such as faces and edges are created explicitly for extrude groups. But, this does not happen at all for lathe groups. See groupmesh.cpp:210

Evil-Spirit, 2015-07-08:
yes, lathe is empty, need to insert some code here https://github.com/whitequark/solvespace/blob/for-upstream/src/group.cpp#L398
i think need to build face only for sketch lines what are perpendicular to the revolution axis. 
is there any other cases?

whitequark, 2015-07-08:
Yeah, face entities should only be built for faces that lie entirely in a plane.

whitequark, 2015-11-03:
Fixed.

closed 2015-11-03 by whitequark