whitequark/solvespace #12: Unicode characters in builtin bitmap font

whitequark, 2015-03-17:


Symbian9, 2015-09-21:
Please implement in internal SolveSpace  UTF8 support for Cyrillic characters and other special characters that presented in character table of "OpenGhost" font files (only  number of symbols, not TTF or geometry from TTF by itself).
http://forum.librecad.org/Opengost-Font-td5710638.html

Maybe change internal font for like this version of Simplex: http://www.azfonts.net/load_font/1196841398_simplex.html

whitequark, 2015-09-21:
Yes, the plan was to take fonts from LibreCAD.

I don't understand what do you mean by "number of symbols"... any UTF-8 characters will be permitted once this is implemented.

Symbian9, 2015-09-21:
Here is SVG version of this font with additional symbols!!!:
1) http://soft9000.com/HersheyShowcase/
2) http://sourceforge.net/projects/hershey9000/

Now, when i try input DEGREES symbol or ANY Cyrillic i have mistake, some times it crash latest build (i386 "v11" for Tahr 32bit from Launchpad).

Symbian9, 2015-09-21:
Is it a way to implement fonts like it it is in LibreCAD? For example make "[fontname].slvs" that like include group where all:
...
Group.name=#font
...
And this group will include all character set based on character codes, for example every workplane "#U+01FF" will include geometry "o", etc.
...
Then SolveSpace will parse every workplane from this group and use it as character!

Think so it need some code, but I'm not programmer :(

whitequark, 2015-09-21:
No, I don't think that's a good idea. What I had in mind is a parser for the font format that LibreCAD uses, since it's simple enough.

Symbian9, 2015-09-21:
Ok, but what is with UTF8 characters input? Is it work for, example any Cyrillic letter? 
When I try both: internal or external TTF font i steal cant input special characters like "°" (the degree symbol) or like "Абвгд"(Cyrillic). Have o empty input or crash SolveSpace (latest build from launchpad).

whitequark, 2015-09-21:
Sure.

I know; SolveSpace currently treats all strings as ASCII; it also has a homegrown TTF parser that doesn't handle anything outside printable ASCII, hence the problem.

Symbian9, 2015-09-21:
If you know Herrshey format (internal Simplex font), then try "reverse" this [*.jhf-->*.lff] convertor:  
http://sourceforge.net/projects/rallaz.u/files/fontconverter/fontconverter.zip/download

whitequark, 2015-09-21:
Yes, that converter should work. Thanks.

Symbian9, 2015-09-22:
So, correct link for [ .jhf-->.lff ] fontconvertor sourcecode is: 
https://github.com/Rallaz/LibreCAD/blob/dwgsupp2/tools/ttf2lff/lfftransform.cpp

Try make "reverseengineering" and make [ .lff-->.jhf ] convertor for SolveSpace.

And, please, add full Unicode character table support as soon, as possible.

Symbian9, 2015-09-22:
Also here is Hershey Fonts convertor for OpenGL apps (PublicDomain)

- info http://sol.gfxile.net/hershey

- sourcecode http://sol.gfxile.net/hershey/hershey.cpp

- all-in-one ZIP with examples http://sol.gfxile.net/hershey/hershey.zip

Maybe, you try make "lff2c.cpp" convertor for SolveSpace (like ttf2c.cpp).

whitequark, 2015-09-27:
ttf2c is for the bitmap fonts; vector fonts are rendered using a separate TTF parser.

Symbian9, 2015-12-06:
Is it some news about Unicode implementation?

whitequark, 2015-12-06:
There's https://github.com/whitequark/solvespace/commits/unicode, but there are some annoying issues related to NameStr.

whitequark, 2015-12-28:
The latest nightly has most of the machinery. Note that the builtin vector font lacks non-ASCII glyphs still, and the TTF parser is kinda broken, but you can try and play with it.

whitequark, 2016-02-13:
Done.

closed 2016-02-13 by whitequark