whitequark/solvespace #13: DXF export is unusable in other CADs

whitequark, 2015-03-17:
There are several major issues with DXF export:

 - [ ] Circles that are projections of lathe edges are exported as line segments (see #4).
 - [ ] Several lines get overlaid on top of each other.
 - [ ] Styles are ignored.

This makes it extremely and unnecessarily hard to process a DWG export in librecad to make a shop drawing.

whitequark, 2015-07-08:
For implementing this I suggest using libdxflib (https://launchpad.net/ubuntu/+source/dxflib/2.5.0.0-2) or libdxfrw (https://github.com/LibreCAD/LibreCAD/tree/master/libraries/libdxfrw). The latter appears to be a fork of the former.

Evil-Spirit, 2015-08-19:
are you still interesting to improve?

whitequark, 2016-02-19:
libdxfrw was integrated.

closed 2016-02-19 by whitequark