whitequark/solvespace #15: "Orthographic projection" rendering mode

whitequark, 2015-03-17:
This mode would have several differences that would help creating shop drawings from solvespace (see also #13). In particular:

 - [ ] Lines that are not edges should not be shown.
 - [ ] Lines that are not directly visible are drawn using dotted lines (should use a builtin style; see #14).

The mode should be reflected in exported drawings as well.

whitequark, 2015-12-30:
To implement this properly requires an OpenCASCADE backend, since a precise curve-surface intersection is needed to hide the inner edges.

whitequark, 2016-02-17:
Actually, we can remove them as pwl while exporting.

whitequark, 2016-04-13:
This is absolutely amazing. I cannot even imagine a better implementation.

@jwesthues Check this out:
[image]

leeoniya, 2016-04-13:
the difference in line weights of visible edges at the hidden intersections is a bit odd...

whitequark, 2016-04-13:
@leeoniya One of them is an outline and the other is a regular solid edge. As far as I'm aware this is how you're actually supposed to draw them, but I'm not completely sure.

whitequark, 2016-04-13:
Actually I might just have been accustomed to CAD-prepared drawings from a CAD that uses the same algorithm. I'll need to get a copy of ISO GPS somewhere... it's really annoying to obtain.

leeoniya, 2016-04-13:
> As far as I'm aware this is how you're actually supposed to draw them

Huh, i'm not sure what's logically different about the hole's edges. What's the reasoning behind this?

whitequark, 2016-04-13:
A solid edge is drawn when the normals are discontiguous between two visible faces, an outline edge is drawn when the normals are discontiguous and one of the faces is invisible.

leeoniya, 2016-04-13:
Never knew that. Looks somewhat disorienting to me as edges morph weights based on orientation. While it may be technically correct for static paper output where this extra information is useful to convey, I find it less than ideal for an interactive display. Perhaps make this option opt-out?

my $0.02.

Thanks for your work on this project, btw; it's great!

whitequark, 2016-04-13:
Well it's not an "option", it's just a consequence of how the algorithm works... dunno how hard it is to exclude these edges. It's worth a look probably.

Re: interactive display, when there's no shaded mesh, I don't see all that much difference in the affordances of the display and paper renders.

whitequark, 2016-04-13:
Oh and in any case, the edges will *always* change weight based on orientation, as any solid edge can be or not be a part of the outline in different orientations.

jwesthues, 2016-04-13:
That looks nice. If I understand correctly, @leeoniya could make all the non-dashed edges appear with the same weight by setting identical line styles for SILHOUETTE and SOLID_EDGE, so there's no need for a specific opt-out option.

Is "orthographic" the right word for this? That usually means something like "parallel projection", which is orthogonal to the question of how hidden lines are displayed. I can't immediately think of a simple alternative, though.

whitequark, 2016-04-13:
> SILHOUETTE and SOLID_EDGE, so there's no need for a specific opt-out option.

Correct. (fwiw I've renamed INVISIBLE to HIDDEN_EDGE and SILHOUETTE to OUTLINE during merging since that seems to be the accepted terms)

> Is "orthographic" the right word for this?

It's absolutely not. I just couldn't think of a better name and I had to name the issue somehow. "Drawing mode" is my most descriptive attempt.

Evil-Spirit, 2016-04-14:
There was actual algorithm to draw outlines with depth-buffer only for visible edges, but for invisible this is not works (we must have solid watertight mesh, if we could invert normals, we get not what we should, so my experiments fails). Actually, if we based on depth buffer, we can get even the pure-outline (wich is not edge for visible-invisible transition edges). But for exported edges we have to run some algorithm which is not fast, so we cannot do it interactive way like the current outline-edges-thing (which works in realtime).

Evil-Spirit, 2016-04-14:
> INVISIBLE to HIDDEN_EDGE
INVISIBLE is better term, because HIDDEN sounds like we are hide it by our intent.

Evil-Spirit, 2016-04-14:
Sorry, actually not
http://help.solidworks.com/2012/English/SolidWorks/sldworks/HIDD_VIEW_SHOW_HIDDEN.htm

whitequark, 2016-04-15:
Ok, I merged all of this except the DXF polyline stuff. I've also changed it so that hidden entities are rendered and exported in the same way as hidden solid edges. Now everything should be straightforward and consistent.

@jwesthues Can you take a look at commits 41436cab20, c7c5d2468 and 9569786? I put quite some effort to make these changes as unobtrustive as possible, but who knows.

closed 2016-04-15 by whitequark

whitequark, 2016-04-15:
@jwesthues To clarify, I'm only asking you to look at the behavior described in the commit messages, there shouldn't be anything so wrong with the implementation that it can't wait until pre-merge review.

jwesthues, 2016-04-15:
At least from a quick review of the commit messages, makes sense to me.

Evil-Spirit, 2016-04-15:
Just leave it here.
[image]
[image]