whitequark/solvespace #16: More 3D operations: loft, fillet, chamfer

whitequark, 2015-03-17:


Evil-Spirit, 2015-07-08:
this is very hard with the current NURBS core :( problem is not to build shell, problems is in booleans which cant process any geometry

whitequark, 2015-07-08:
Yeah, I've learned about this already. How problematic would it be to use a different NURBS core?

Evil-Spirit, 2015-07-08:
we are trying at the momnet. but we still not found good nurbs library. there is OpenCASCADE - the big and the fat.

whitequark, 2015-07-08:
Yeah, OpenCASCADE is horrible. I'd really rather avoid it. Have you seen http://www.sintef.no/Projectweb/Geometry-Toolkits/SISL/? It looks exactly like what we need.

Evil-Spirit, 2015-07-18:
no. no. no. this is shit.

whitequark, 2015-07-18:
ok. I guess it is worth trying to integrate OCE. hopefully this can be done so that OCE does not infest the rest of codebase.

jonnycowboy, 2015-11-30:
What about integrating Verb Nurbs instead?  Very power, easy to use, etc...

Evil-Spirit, 2015-12-11:
as i understand, it have no completed booleans, topo branch probably unstable

whitequark, 2015-12-30:
We need OpenCASCADE first for this. Looks like there's no other option.

closed 2015-12-30 by whitequark

agryson, 2016-07-22:
bump
Is there any more news on this?
Is it planned for after 3.0?

whitequark, 2016-07-22:
There are no current plans.

rzvc, 2017-06-21:
Any updates on this?

If this will be done it will take SolveSpace to a new level. I guess it will also lay down the path for extruded bodies that follow a custom path which will enable a lot more - threads for example.

whitequark, 2017-06-21:
There are no current plans. Do not expect this to happen.