whitequark/solvespace #17: Allow specifying a workplane by a point and two normals, or a point, line and a normal

whitequark, 2015-03-18:
These come up quite often, and otherwise require adding awkward construction entities with tangent/parallel/perpendicular constraints.

whitequark, 2015-12-30:
Ah, this wouldn't really work, as some combinations are ambiguous.

whitequark, 2015-12-30:
Ah, no, that already happens with "point and two line segments", if you put them all on the same line. The sketch just becomes very unstable.

whitequark, 2015-12-30:
This is useful to start working on a side of an extrusion, especially with arcs; you already have a normal, so you add a point on the arc and draw a tangent line through it.

Evil-Spirit, 2016-04-02:
I dont see the usecase where we can't use the (already existent) axis of revolution or segment from extrusion of center-point of circle/arc to use it instead of normal. If we will introduce it, we will break file format backward-compatability.

whitequark, 2016-04-02:
We can't use a line segment and a normal at the same time, and we also cannot use two line segments that don't go through the same point.

Evil-Spirit, 2016-04-02:
Actually we can use two line segments which is actually don't go through the same point (we take only the directions) and splecify any point for workplane origin (i've just tested it).

whitequark, 2016-04-02:
Does that already work or does it require code modification? If latter, yes, this is fine by me.

Evil-Spirit, 2016-04-02:
This already work without any modifications.

whitequark, 2016-04-02:
Oh, great. Then just the help text needs updating--I'll do it.

whitequark, 2016-04-08:
Oh haha, I cannot read. It never said that the *lines* need to go through the point.

closed 2016-04-08 by whitequark