whitequark/solvespace #28: Build fails on OS X 10.10.3

lhagan, 2015-06-13:
Build failed at the linking stage on Yosemite (OS X 10.10.3). Should this have worked, or is Yosemite not yet supported?

```
Linking CXX executable solvespace.app/Contents/MacOS/solvespace
Undefined symbols for architecture x86_64:
  "std::string::c_str() const", referenced from:
      SolveSpace::SolveSpaceUI::MenuFile(int) in solvespace.cpp.o
  "std::string::empty() const", referenced from:
      SolveSpace::RefreshRecentMenu(int, int) in cocoamain.mm.o
      SolveSpace::SaveFileYesNoCancel() in cocoamain.mm.o
  "std::string::rfind(char, unsigned long) const", referenced from:
      Extname(std::string) in solvespace.cpp.o
  "std::string::length() const", referenced from:
      Extname(std::string) in solvespace.cpp.o
  "std::string::substr(unsigned long, unsigned long) const", referenced from:
      Extname(std::string) in solvespace.cpp.o
  "std::allocator<char>::allocator()", referenced from:
      SolveSpace::RefreshRecentMenu(int, int) in cocoamain.mm.o
      SolveSpace::SaveFileYesNoCancel() in cocoamain.mm.o
      SolveSpace::SolveSpaceUI::MenuFile(int) in solvespace.cpp.o
      Extname(std::string) in solvespace.cpp.o
  "std::allocator<char>::~allocator()", referenced from:
      SolveSpace::RefreshRecentMenu(int, int) in cocoamain.mm.o
      SolveSpace::SaveFileYesNoCancel() in cocoamain.mm.o
      SolveSpace::SolveSpaceUI::MenuFile(int) in solvespace.cpp.o
      Extname(std::string) in solvespace.cpp.o
  "std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)", referenced from:
      SolveSpace::RefreshRecentMenu(int, int) in cocoamain.mm.o
      SolveSpace::SaveFileYesNoCancel() in cocoamain.mm.o
      SolveSpace::SolveSpaceUI::MenuFile(int) in solvespace.cpp.o
      Extname(std::string) in solvespace.cpp.o
  "std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()", referenced from:
      SolveSpace::RefreshRecentMenu(int, int) in cocoamain.mm.o
      SolveSpace::SaveFileYesNoCancel() in cocoamain.mm.o
      SolveSpace::SolveSpaceUI::MenuFile(int) in solvespace.cpp.o
  "std::ios_base::Init::Init()", referenced from:
      ___cxx_global_var_init in cocoamain.mm.o
  "std::ios_base::Init::~Init()", referenced from:
      ___cxx_global_var_init in cocoamain.mm.o
  "std::_Rb_tree_decrement(std::_Rb_tree_node_base*)", referenced from:
      std::_Rb_tree_iterator<std::pair<int const, NSMenuItem* __strong> >::operator--() in cocoamain.mm.o
  "std::_Rb_tree_increment(std::_Rb_tree_node_base*)", referenced from:
      std::_Rb_tree_iterator<std::pair<int const, NSMenuItem* __strong> >::operator++() in cocoamain.mm.o
  "std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&)", referenced from:
      std::_Rb_tree<int, std::pair<int const, NSMenuItem* __strong>, std::_Select1st<std::pair<int const, NSMenuItem* __strong> >, std::less<int>, std::allocator<std::pair<int const, NSMenuItem* __strong> > >::_M_insert(std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::pair<int const, NSMenuItem* __strong> const&) in cocoamain.mm.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/solvespace.app/Contents/MacOS/solvespace] Error 1
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
make: *** [all] Error 2
```

whitequark, 2015-06-13:
Looks like a libc mismatch. Try following instructions in https://stackoverflow.com/questions/11972603/cmake-clang-and-cv11-on-os-x-10-8; I don't have a Yosemite VM and cannot test.

lhagan, 2015-06-13:
Thanks, adding this to `CMakeLists.txt` solved it:

    set(CMAKE_CXX_FLAGS "-stdlib=libstdc++")

However, the resulting app is missing all icons. The binary from GitHub works fine. There were some warnings from `cmake`; not sure if any of this could be the cause:

```
-- Checking for perl module GD
-- Checking for perl module GD - failed
-- Could NOT find PerlModules (missing:  PERLMODULES_GD_MODULE) 
-- Perl with GD not found; icons will not be regenerated if modified
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   slvs

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/lhagan/Documents/Projects/@Temp/solvespace/cbuild
```

whitequark, 2015-06-13:
Try installing perl-gd (it should be in brew). This is a bug, but it will solve the problem for you shortly.

lhagan, 2015-06-13:
I was able to get rid of the perl-gd error, but it didn't fix the missing icon issue. In case it helps someone else, here's one way to install perl-gd if you're not used to working with `perl` and don't want to use `sudo`:

```
# install libgd with homebrew
brew install libgd

# install cpanm into home folder (http://stackoverflow.com/a/8313215/784195)
curl -L http://cpanmin.us | perl - -l ~/Developer/perl5 App::cpanminus local::lib
eval `perl -I ~/Developer/perl5/lib/perl5 -Mlocal::lib=~/Developer/perl5`
echo 'eval `perl -I ~/Developer/perl5/lib/perl5 -Mlocal::lib=~/Developer/perl5`' >> ~/.profile

# install perl-gd
cpanm GD
```

whitequark, 2015-06-13:
Are you building the for-upstream branch?

lhagan, 2015-06-13:
I was on `master`. Switching to `for-upstream` did it. Thanks again!

closed 2015-06-14 by whitequark

tmpvar, 2015-10-13:
`perl-gd` is not in brew

```
solvespace ~ build (for-upstream)‣ brew install perl-gd
Error: No available formula for perl-gd
==> Searching formulae...
==> Searching taps...
```

I'm seeing the same issue:

```
solvespace ~ build (for-upstream)‣ cmake ../
-- Checking for perl module GD
-- Checking for perl module GD - failed
-- Could NOT find PerlModules (missing:  PERLMODULES_GD_MODULE)
-- Perl with GD not found; icons will not be regenerated if modified
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   slvs

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/tmpvar/work/cad/solvespace/build
```

whitequark, 2015-10-13:
@tmpvar, perl-gd is not a hard dependency. The build will succeed without it.

tmpvar, 2015-10-13:
```
solvespace ~ build (for-upstream)‣ make
Scanning dependencies of target slvs
[  1%] Building CXX object src/CMakeFiles/slvs.dir/util.cpp.o
[  3%] Building CXX object src/CMakeFiles/slvs.dir/entity.cpp.o
[  4%] Building CXX object src/CMakeFiles/slvs.dir/expr.cpp.o
/Users/tmpvar/work/cad/solvespace/src/expr.cpp:528:9: warning: 'this' pointer cannot be null in well-defined C++ code; pointer may be assumed to always convert to true [-Wundefined-bool-conversion]
    if(!this) return "0";
       ~^~~~
1 warning generated.
[  6%] Building CXX object src/CMakeFiles/slvs.dir/constraint.cpp.o
[  7%] Building CXX object src/CMakeFiles/slvs.dir/constrainteq.cpp.o
[  9%] Building CXX object src/CMakeFiles/slvs.dir/system.cpp.o
[ 10%] Building CXX object src/CMakeFiles/slvs.dir/unix/unixutil.cpp.o
[ 12%] Building CXX object src/CMakeFiles/slvs.dir/lib.cpp.o
[ 14%] Linking CXX shared library libslvs.dylib
[ 14%] Built target slvs
[ 15%] Bundling shared library /usr/local/lib/libpng.dylib
[ 17%] Building Interface Builder file cocoa/MainMenu.xib
[ 18%] Building Interface Builder file cocoa/SaveFormatAccessory.xib
/* com.apple.ibtool.document.notices */
/Users/tmpvar/work/cad/solvespace/src/cocoa/SaveFormatAccessory.xib:z9Z-cA-QIW: note: View is clipping its content [13]
[ 20%] Building icon set cocoa/AppIcon.iconset
/Users/tmpvar/work/cad/solvespace/src/cocoa/AppIcon.iconset:error: Iconset contains no image resources.
/Users/tmpvar/work/cad/solvespace/src/cocoa/AppIcon.iconset:error: Failed to generate ICNS.
make[2]: *** [src/solvespace.app/Contents/Resources/AppIcon.icns] Error 1
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
make: *** [all] Error 2
```

whitequark, 2015-10-13:
Oh. Well, the `iconutil` invocation succeeds (with warnings) on XCode 5.1.1. What version of XCode are you using?

tmpvar, 2015-10-13:
7.0.1 (7A1001)

whitequark, 2015-10-13:
Well, I can't update to 7.0.1 because my OS X is too old, also update at all because the App Store gives me some opaque error, or install a newer OS X because the Yosemite or El Capitan installation media does not appear to include any bootloader my VM can recognize, and I have no Apple hardware. So it seems you would have to figure this out on your own.

tmpvar, 2015-10-13:
will do, just dropping some breadcrumbs :)

whitequark, 2015-11-04:
@tmpvar any update on this?

whitequark, 2015-11-04:
@tmpvar by the way, I've removed the perl dependency entirely :)

tmpvar, 2015-11-04:
I'll try again in a few hours

MartinBloedorn, 2016-01-20:
Any updates? 

I'm hitting the same error under  OSX 10.11. On the `make` step, I also get the error: 

    [ 23%] Building icon set cocoa/AppIcon.iconset
    /Users/martin/Git/solvespace/src/cocoa/AppIcon.iconset:error: Iconset contains no image resources.
    /Users/martin/Git/solvespace/src/cocoa/AppIcon.iconset:error: Failed to generate ICNS.
    make[2]: *** [src/solvespace.app/Contents/Resources/AppIcon.icns] Error 1
    make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
    make: *** [all] Error 2

Any ideas? :)

whitequark, 2016-01-20:
@MartinBloedorn That is a known bug. Unfortunately I have no clue how to fix it as I do not have (and never had) access to any Apple hardware and the way I run OS X in a VM doesn't really work for 10.9 and later. You can use binaries from https://github.com/whitequark/solvespace/releases if you'd like something quick.

MartinBloedorn, 2016-01-20:
@whitequark Thanks for the prompt answer :) Binaries worked ok; thanks!