whitequark/solvespace #34: Make daily builds as PortableLinuxApp (i386 and x86_64)

Symbian9, 2015-09-21:
Think SolveSpace must be "portable" like "dmg" or "exe", without any dependency.
There is a way -  "PortableLinuxApps" are apps that run on most 32-bit or 64- bit Linux-based operating systems, such as Ubuntu, Fedora, debian and derivatives. 
They are bundled into self-containing files: 1 app = 1 file (*.run, ELF).
http://portablelinuxapps.org/

Please, add this as part of daily(nightly) builds for resolve dependency issues for Linux users!

P.S.: I use TahrPup - PuppyLinux (~200Mb only!), based on Trusty 14.04. In PuppyLinux they have "SFS package" for portability apps.
Now previos build "solvespace_2.1_v10~trusty_i386" added - http://distro.ibiblio.org/puppylinux/pet_packages-tahr/solvespace-2.1.sfs.

whitequark, 2015-09-21:
No, that's a terrible solution. The proper way to handle this is to use SUSE's [Open Build Service](https://build.opensuse.org/) and build native packages for all interesting distributions. I don't have time for that, though.

closed 2015-09-21 by whitequark

Symbian9, 2015-09-21:
But, is it a way make daily/nightly builds in style like Blender3D, when all shared libs included and it run from unzipped folder?
For example, how install this for offline PC with "LinuxMint 15" ?

whitequark, 2015-09-21:
Use [apt-offline](http://apt-offline.alioth.debian.org/) for installing anything on offline PCs.

Symbian9, 2015-09-21:
apt-offline need registering... I only wont "every day fresh" version of SolveSpace like unzip and run, without any carry of dependency, but have not win or mac.

Offtopic: Maybe its real make port for KolibryOS (assembler os)?

whitequark, 2015-09-21:
You have the source. You are free to compile SolveSpace however you want.

And no.