whitequark/solvespace #36: AC3D (.ac) exporter

Symbian9, 2015-10-09:
Please, add AC3D format for export that will work ou tof the box. 

https://en.wikipedia.org/wiki/AC3D

Here is a python addon for Blender (with comments in passing code)
https://github.com/majic79/Blender-AC3D

And here is few convertors with opensource writen on C / C++
http://texel3d.free.fr/projets/ani2pov/
http://www.inivis.com/forum/showthread.php?t=1923
http://www.programdoc.com/312_1551_1.htm

whitequark, 2015-10-09:
I am not interested in supporting this format.

Symbian9, 2015-10-10:
So, maybe you have plane how implement some simple scripting for easy way make plug-in for SolveSpace. It would be greate add some possibility to easy way add parser rules without need now C++ and recompil your program if need only add this future. What You think?

whitequark, 2015-10-10:
Not happening any time soon.

closed 2015-10-10 by whitequark