whitequark/solvespace #37: Three.js mesh export rule for CanvasRenderer ( for non-WebGL)

Symbian9, 2015-10-10:
Thanks, that it have exporter for WebGLRenderer of Three.js
https://github.com/whitequark/solvespace/issues/33

But can you make one more rule for this exporter, that will be preconfigurated for older hardware that steal not have WebGL implemented.
Think you can add CanvasRenderer rules for presented exporter html code as comment. Than user can switch quickly uncomment it in html if need CanvasRender for non-WebGL.

whitequark, 2015-10-10:
Implementing software 3D rendering in JavaScript is a non-starter. Besides, I do not care about such old hardware.

closed 2015-10-10 by whitequark

whitequark, 2015-11-01:
Hang on, I've just realized that CanvasRenderer is already implemented as a part of three.js.

Anyway, I've just checked and its rendering results are substantially inferior to proper WebGL. Anyone who actually wants these inferior results can change WebGLRenderer to CanvasRenderer directly, it is not hard.