whitequark/solvespace #39: Couldn't compile on archlinux

tgirod, 2015-10-17:
Hi. It wasn't compiling because the option ``-std=c++11`` was missing to GCC's call. With this patch it worked.

~~~
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 796ef81..8f4ab3d 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -41,8 +41,9 @@ if(WIN32)
         -DWIN32_LEAN_AND_MEAN=1)
 endif()
 
-if((CMAKE_CXX_PLATFORM_ID STREQUAL "Linux") AND CMAKE_COMPILER_IS_GNUCC)
+if((CMAKE_SYSTEM_NAME MATCHES "Linux") AND CMAKE_COMPILER_IS_GNUCC)
     set(CMAKE_EXE_LINKER_FLAGS "-Wl,--as-needed ${CMAKE_EXE_LINKER_FLAGS}")
+    set(CMAKE_CXX_FLAGS "-std=c++11")
 endif()
 
 if(MINGW)
~~~

whitequark, 2015-10-19:
Does this not work for you? https://github.com/whitequark/solvespace/commit/cbe1730b7ad8f2227126d365cb3b64813ab9ef53

tgirod, 2015-10-19:
It worked.

closed 2015-10-19 by whitequark