whitequark/solvespace #40: arcs and constraints

tgirod, 2015-10-18:
Hi! I'm having troubles applying constraints on arcs and circles. For example:

1. Draw a segment
2. Draw an arc from one end of the segment to the other end
3. select the center of the arc and constrain it to the middle of the segment
4. move the shape around. It works pretty well if you don't move it around too quickly
5. add a distance constrain on the segment

Here, I see two things: the constrain generally fails, I can make it work by gently nudging the shape, but then it gets very fragile - move just a bit too much, too quickly and the constraints will fail, causing you to break the shape by dislodging a segment or a point.

Am I doing something wrong?

whitequark, 2015-10-18:
Use a collinearity constraint instead

tgirod, 2015-10-19:
Nice, it worked! thanks a lot.

Can you explain me why it behaves like this ? I think my constraints were valid, but broke anyway. Is it because the constraints are solved numerically, thus producing a result that might not be exactly right (I mean, if the correct result is a real or a quotient) ?

whitequark, 2015-10-19:
Actually, now I try to reproduce it and my own advice does not work. In fact I don't even know how collinearity constraint would be useful here--I have originally misread your report.

Can you attach a slvs file?

tgirod, 2015-10-19:
https://gist.github.com/tgirod/7475387a64dd7d7e97bd#file-bug-slvs (I hope the copy-paste worked as intended)

Anyway, the idea is, if I draw an arc A between the two extremities of a segment S, A's center has to be somewhere on a line perpendicular to S, intersecting S in its middle.

So if I force A's center to the middle of S, it is valid. It is also valid to force A's center to be colinear to S, as it should theoretically lead to the same unique solution. Moreover, it should work with any dimension of S.

What I see is that the more constraints I add, the more fragile the shape becomes - Moving it around causes failures of the constraint solver.

From a UI perspective, it is very annoying because if the constraints fail when moving the shape, the movement is not stoped right now - instead, it continues a bit only with the selected element, thus breaking the shape and causing another error.

I hope what I say is clear ...

whitequark, 2015-10-19:
This has actually nothing to do with the distance constraint, the issue is just that adding two coincident constraints and a midpoint constraint overconstrains the sketch. This is a deliberate design decision in the solver with the goal being the clarity of the sketch: only the constraints that are necessary and sufficient are considered valid.

I do think a setting to toggle this behavior would be useful and will likely implement one; see also https://github.com/jwesthues/solvespace/issues/2#issuecomment-96236774.

tgirod, 2015-10-19:
OK, I think I get it, but something feels backward. Please bear with me for a moment. 

- pts-coincident
- pts-coincident
- pt-pt-distance
- pt-on-line

With those constraints, the solver passes and give me 3 DOF. I get it for the two extremities of the segment, but how is the center of the arc still considered a DOF ? Technically it could move along the segment, but it has to be equidistant to the extremities, so the segment's midpoint is the only valid solution here.

But then, if I replace ``pt-on-line`` with ``at-midpoint`` and explicitely remove this DOF, the sketch is overconstrained ?

whitequark, 2015-10-20:
I think the reason is that only the depths of freedom that are analytically, not numerically, removed are counted in the DOF number. @evilspirit may offer some more enlightenment here.

whitequark, 2015-10-20:
I think the reason is that only the depths of freedom that are analytically, not numerically, removed are counted in the DOF number. @evil-spirit may offer some more enlightenment here.

Evil-Spirit, 2015-10-24:
the nature of the arc is point1-center and point2-center distance equality. when you constrained it to the segment, you already get something like at-mid-point constraint. so, it becomes overconstrained when at-mid-point specified in this case. but solvespace`s solver uses some method to compute dof number for the sketch, but this analys based on the rank calculating for the jacobian, so this probably can`t catch cases like this and causes overconstrained system solving. typically, this is not big problem what we solve overconstrained system with equations of the same nature, but if we have to solve equations of the different types but the same meaning, we probably get conflicting partial derivatives and bad convergence.

Evil-Spirit, 2015-11-24:
i`ve catch it!
there is some interesting behaviour: dof number and overconstraining is calculated before the solving, and rank test says it'ok: rank == mat.m (https://github.com/whitequark/solvespace/blob/for-upstream/src/system.cpp#L452). but after solving things becomes linear-dependent, so after solving rank test is not ok(even when solving was sucessful), but we can't catch it and report as error because we don't want to make FindWhichToRemoveToFixJacobian() while dragging. so, i think it is good solution to check rank after solving when we want to create new constraint and FindWhichToRemoveToFixJacobian().

whitequark, 2015-11-24:
Looks good to me.

whitequark, 2016-01-22:
Fixed in c0114440.

closed 2016-01-22 by whitequark