whitequark/solvespace #41: inconsistent system: impossible or redundant ?

tgirod, 2015-10-20:
When working on a sketch, if I get something wrong with my constraints the solver tells me my constraints are inconsistent. Unfortunately, it doesn't say if I'm trying to do something impossible, or if some of my constraints are redundant.

Example: 

1. draw a triangle and add two "equal length" constraints
2. add a third "equal length" (redundant) constraint -> ``SOLVE FAILED! inconsistent system``
3. add a "perpendicular" (impossible) constraint -> ``SOLVE FAILED! inconsistent system``

I think it would help a lot (at least for a beginner like me) to have a more explicit error. As it is, it can be very misleading.

whitequark, 2015-10-20:
Good point

Evil-Spirit, 2015-10-25:
please, provide an example.

whitequark, 2015-10-25:
Examples:

[fail-redundant.slvs | uploaded via ZenHub](https://files.zenhub.io/562c7e17e7d5851e348a435f)

[ok.slvs | uploaded via ZenHub](https://files.zenhub.io/562c7e17e7d5851e348a4360)

[fail-impossible.slvs | uploaded via ZenHub](https://files.zenhub.io/562c7e17e7d5851e348a435e)

Evil-Spirit, 2015-10-28:
I think we can differentiate some cases when we have solver fail or overconstrained equation system & give corresponding advice for the user.

whitequark, 2015-10-28:
That would be wonderful, especially if it is not a lot of work.

whitequark, 2016-01-22:
Fixed in 55cde18c5a483ba69bc509717a1f515626871c77.

closed 2016-01-22 by whitequark