whitequark/solvespace #46: undelete sketches at startup is not working

tgirod, 2015-11-18:
I made the mistake of editing a sketch A on which another sketch B is dependant.

When opening B, I get the following message:

```Additional sketch elements were deleted, because
they depend on the element that was just deleted
explicitly. These include: 
     35 requests
     99 constraints
     7 groups

Choose Edit -> Undo to undelete all elements.```

The undo function is grayed out, so it is not possible to undelete all elements.

I'm running solvespace v2.1

whitequark, 2015-11-18:
Yes. It's not possible to do that. Instead, restore A from your version control system and reopen B without previously saving it.

tgirod, 2015-11-18:
Maybe the error message should be adapted accordingly. If only I had used a VCS. thanks for the quick answer.

whitequark, 2015-11-18:
I agree.

tgirod, 2015-11-20:
OK, I think I get it. When something is changed upstream, anything orphaned downstream will be deleted. Working in a single file, I can undo this, but not with multiple files. It seems rather difficult to guess what will be deleted and what will not - looks like I lost a game of russian roulette here.

Would it make sense not to destroy orphans, but instead replace the missing points with new unconstrained points set to the former coordinates - with a warning message ?

That way it would be possible fix a sketch upstream without the risk of breaking downstream work, which could be adapted rather than recreated from scratch.

Well I guess there would be a few corner cases like: what happens if I destroy a facet which is used as a workplane downstream ?

whitequark, 2015-11-20:
That would be a very complex change and I don't think it improves the experience enough.

whitequark, 2016-01-11:
Done

whitequark, 2016-01-11:
(Specifically, the error message was adjusted.)

closed 2016-01-11 by whitequark