whitequark/solvespace #47: Not supported new GLEW libs

Symbian9, 2015-11-27:
Found problem with latest nightly relises, that now cant work with lib "libglew1.10"(newer), and work only with "libglew1.6" (older).
Tested on Trusty.

But previos "solvespace_2.1~trusty11_amd64.deb" steal normally work with "libglew1.10"(newer).

whitequark, 2015-11-27:
Define "can't work". I only have libglew1.10, and it runs fine.

Symbian9, 2015-12-01:
I use some distr of LinuxMint x86_64.
My PC is only offline, so I download all packages and install them manually in offline mode.
Firstly I install latest version of SolveSpace from launchpad.net (relised 19 Oct 2015).
./DEBIAN/control
-----
Version: 2.1~trusty11
Depends: ..., libglew1.10  (>= 1.10.0), ...
-----
As you can see it was need libglew1.10. In my distribution it not presented from start, so I install it manualy.

Later I download latest Nightly build from github (relised 3 Nov 2015).
./DEBIAN/control
-----
Version: 2.1 #why You not add date or build number?
Depends: ..., libglew1.6 (>= 1.6.0), ...
------
Then I remove "solvespace_2.1~11", and When I try delite previouse and install latest it tell me that I must install "libglew1.6", but why need install previos if I already have higher version?
Anyway, later I download "libglew1.6" and install it too (so at the moment I have both 1.6.0-4 and 1.10.0-3 installed).
Only after that I can install and run latest nightly build from github.

So what the problem here?

whitequark, 2015-12-01:
This isn't really a problem. The Launchpad buildbots use libglew1.10, and my Travis buildbots use libglew1.6. Neither of these choices is incorrect because you can install both ABI versions of the libraries at once.

closed 2015-12-01 by whitequark