whitequark/solvespace #50: [BUG] Crash when open any assembly examples

Symbian9, 2015-12-13:
I use latest SolveSpace nightly relise from github (where "libslvs" for this build?), but when it try open any "asm.slvs" distributed on official site in Examples section and some examples on github (I unpack all files for assembly in same folder), it crashed with next error in bash:
<code>oops at line 706, file /home/travis/build/whitequark/solvespace/src/entity.cpp

Aborted</code>

P.S.: previouse build from Launchpad ( ver.2.1~trusty11 for solvespace and libslvs) open it normally.

whitequark, 2015-12-13:
cc @Evil-Spirit

Evil-Spirit, 2015-12-14:
fixed at https://github.com/Evil-Spirit/solvespace/commit/02be1d4ca4438ea405141a18c77504524b6707d3

Symbian9, 2015-12-19:
Please, make new binary (solvespace&libslvs) from latest sources with bug fixes for Ubuntu 14. 04(x86 & x86_64) on Launchpad! Thanks.

closed 2015-12-25 by whitequark