whitequark/solvespace #51: oops at line 644, file /src/drawentity.cpp

aam2015, 2015-12-24:
Built version 2.1 from source today (Dec 23 2015) on an LMDE 2 64-bit system. Importing any of the parts from the assembly example (http://solvespace.com/dl/box-parts.zip) causes an immediate abort with the above error.

Symbian9, 2015-12-24:
@aam2015
Did you read previous issue?
https://github.com/whitequark/solvespace/issues/50

Anyway can you also:

1) ... upload your build for testing

2) ... make fresh build from @Evil-Spirit repo with fixed issues https://github.com/Evil-Spirit/solvespace/commit/02be1d4ca4438ea405141a18c77504524b6707d3

whitequark, 2015-12-25:
The crash was fixed but there's now some sign error...

whitequark, 2015-12-28:
Fixed in latest nightly

aam2015, 2016-01-04:
The latest fix has created a new issue with all the files above. For example, the normals for the slots in base.slvs are now facing in the opposite direction. This obviously causes the slots not to appear in the base.

whitequark, 2016-01-04:
@aam2015 Hm, correct. @Evil-Spirit, would you please fix this?

whitequark, 2016-01-11:
This was fixed, however, the entire feature (Workplane on Face) was temporarily rolled back until we have a testsuite. Latest nightly should be able to successfully open all files that exhibited bugs.

aam2015, 2016-01-11:
I tried the latest nightly (both building from source and the deb) and neither of them seem to have resolved the latest issue. The normals are all still facing in the opposite direction.

whitequark, 2016-01-11:
That's extremely odd. I've bisected your problem and the commit which caused it has nothing to do with normals. Regardless, I will address this soon.

whitequark, 2016-01-11:
@aam2015 I'm really sorry for all this breakage and back-and-forth. I've fixed this in https://github.com/whitequark/solvespace/commit/9ce4699753eacaa6033ffc00e85d41080d81bc69 and pushed a new nightly.

closed 2016-01-11 by whitequark

aam2015, 2016-01-12:
@whitequark Thanks for taking the time to fix this.

Evil-Spirit, 2016-01-12:
I think it is floating bug and it seems it not has been fixed.

whitequark, 2016-01-12:
Hm, doubt it--my bisecting pointed to a valid cause & I can't reproduce it anywhere anymore.