whitequark/solvespace #52: Point on bezier

whitequark, 2016-01-11:
See http://solvespace.com/forum.pl?action=viewthread&parent=852

Evil-Spirit, 2016-01-22:
I think, it's easy, if we consider:

> representation of the point by its parameter along the curve instead of (u, v) in the plane. 
this is really the most elegant solution, and one that makes good use of the symbolic algebra system underlying SolveSpace's constraint solver, but more work.

Make this is rather simple, if we want to use it for a cubic spline segment alone (4pts), but for whole spline(up to 15 pts) with G2-continuity, we have to make somethig trickly, like adding new type of Expr.
Something like Expr::FUNC will be great.
```c++
struct ExprFunc {
    virtual void value(double x) const = 0;
    virtual void derivative(double x) const = 0;
};

class Expr {
    ...
    enum {
        ...
        FUNC_VALUE,
        FUNC_DERIVATIVE,
        ...
    };
    ...
    union {
        double  v;
        hParam  parh;
        Param  *parp;
        Expr    *b;
        ExprFunc *f;
    };
    ...
};

```

Evil-Spirit, 2016-05-04:
```c++
struct ExprFunc {
    virtual const char *name() const = 0;
    virtual void value(double x) const = 0;
    virtual ExprFunc *derivative() const = 0;
};

class Expr {
    ...
    enum {
        ...
        FUNC
        ...
    };
    ...
    union {
        double  v;
        hParam  parh;
        Param  *parp;
        Expr    *b;
        ExprFunc *f;
    };
    ...
};
```

Evil-Spirit, 2016-05-04:
@jwesthues,
```c++
struct ExprFunc {
    virtual const char *name() const = 0;
    virtual void value(double x) const = 0;
    virtual ExprFunc *derivative() const = 0;
};

class Expr {
    ...
    enum {
        ...
        FUNC
        ...
    };
    ...
    union {
        double  v;
        hParam  parh;
        Param  *parp;
        Expr    *b;
        ExprFunc *f;
    };
    ...
};
```

Evil-Spirit, 2016-05-21:
@jwesthues, need you opinion about this. How do you think: can it work?

Evil-Spirit, 2016-05-21:
@jwesthues,
We can use this to represent picewise functions like bezier_x(t), bezier_y(t), bezier_z(t).
```
struct ExprBezierX : public ExprFunc {
    ExprBezeirX(Entity *bezier) { this->bezier = bezier; }
    void value(double x) cosnt {
        int piece = x / bezier->numPoints;
        // return x value for piece and piece + 1 points
    }
    virtual ExprFunc *derivative(Param *p) const {
        return new ExprBezierDerivX(this->bezier, p);
    }
   
}
```

jwesthues, 2016-05-22:
@Evil-Spirit, I don't think anything that you've posted addresses the difficult part of piecewise functions, computing the derivatives of the point's coordinates with respect to the spline control points. That's messy, because all the pieces are linked through the continuity criteria. The additional layer of abstraction that you propose above doesn't seem like it makes that much easier.

Evil-Spirit, 2016-05-22:
@jwesthues, what about this:
```
struct ExprFunc {
    virtual const char *name() const = 0;
    virtual void value(double x) const = 0;
    virtual Expr *valueExpr(double x) const = 0;
    virtual ExprFunc *derivative() const = 0;
};

struct ExprBezierX : public ExprFunc {
    Param *t;
    Entity *bezier;

    ExprBezeirX(Entity *bezier) {
        t = new Param(0.0);
        this->bezier = bezier;
    }
    
    void value(double x) const {
        t->val = fmod(x, 1.0 / bezier->numPoints) * bezier->numPoints;
        // ofc we can cash Expr
        return valueExpr(x)->Eval();
    }
    
    virtual Expr *valueExpr(double x) const {
        int piece = x / bezier->numPoints;
        // return x_value(t) for piece and piece + 1 points
    }

    virtual ExprFunc *derivative(Param *p) const {
        return new ExprBezierDeriv(this, p);
    }
};

struct ExprBezierDeriv : public ExprFunc {
    ExprFunc *func;
    Param *param;

    ExprBezierDeriv(ExprFunc *fn, Param *p) {
        func = fn;
        param = p;
    }
    
    void value(double x) const {
        t->val = fmod(x, 1.0 / bezier->numPoints) * bezier->numPoints;
        return valueExpr(x)->Eval();
    }
    
    virtual Expr *valueExpr(double x) const {
        return fn->valueExpr(x)->PartialWrt(param);
    }

    virtual ExprFunc *derivative(Param *p) const {
        return new ExprBezierDeriv(this, p);
    }
};
```

jwesthues, 2016-05-22:
I still don't see how this addresses the hard problem. Let's say that you have a two-piece spline, and a point-on-spline identified by t along that spline. How do you propose to compute the derivative of that point's x coordinate with respect to the intermediate on-curve control point's x coordinate?

whitequark, 2016-11-27:
@Evil-Spirit This is low priority, but I'd like to see you spend a small amount of time doing a prototype, and see if it works out.

Evil-Spirit, 2016-12-10:
@whitequark, I've looked into this and found how the current spline system works: the final computation of the spline parameters goes inside ComputeInterpolatingSpline. It performs computation of spline parameters with g2-contiunity without involving the solver, and we can't use this as parameters inside solver, so the pt-on-spline equations can't drive the spline derivative parametes. Without this, we can't do  two-way interactions for pt-on-spline, we can't perform correct dof analysing and so on. So, for doing this we must:
* get rid of ComputeInterpolatingSpline (and BandedMatrix), move all solving of the spline continuity inside solver with adding correspondent number of equations.
* I think there is no needs in supporting more than 4 points for SPLINE request (since this is just cubic spline and can be fully defined by using 4pts).
* The 1st or 2nd derivative stitching can be performed through using the unified TANGENT constraint based on CUBIC_LINE_TANGENT+CURVE_CURVE_TANGENT with supporting of g2 continuity when one of arguments(or both?) is cubic spline (since 2nd derivative makes sense only for cubic splines)
* The spline tool will create new spline(4pts based) for each new point except the second and perform automatic TANGENT constraint apply for support an old behaviour
* The splines requests for old files must be automatically converted into new representation. This is something like complex task.
* For doing all the universal way, we must invent pt-on-polyline (polyprimitive, polyentity). This is not limiting us using only stitched cubic splines, we get ability to use any TANGENT stitched primitives because of continuity of the derivatives.
* for pt-on-polyprimitive we can use some piecewise function representation with unified t parameter spach using the hack with zeroing the terms when parameter t is outside the range

@jwesthues, can you comment this?
I think this is huge amount of work, and the current architecture of SolveSpace don't allow us to do such things in easy way.

whitequark, 2016-12-10:
@Evil-Spirit To rephrase. There are two problems, right?

  1. The CUBIC entity is doing more work than other entities, and it would be more accurately represented by n entities, for n additional control points. 
  2. If CUBIC only did as much work as other entities then we could easily have a point-on-cubic constraint, but that wouldn't do what's intuitively expected because beyond the endpoints the point would have just resided on some curve, continued from the spline. So we would have to have a way to make constraints that transparently span multiple entities.

(1) seems perhaps conceptually easily doable if with a large amount of legwork, to translate old files. (2) seems hard, both in the solver, and in the UI.

In any case, it seems that there is a clear direction for implementing this, but no clear understanding of how long it will take, nor a benefit that's worth the cost. I am therefore rejecting this feature.

closed 2016-12-10 by whitequark

Evil-Spirit, 2016-12-10:
@whitequark, yes, you are right. 
I think it is much simplier to make absolutely different new entity (cubic with 4pts) and leave old cubic spline for legacy files for some period of supporting time.

whitequark, 2016-12-10:
> leave old cubic spline for legacy files for some period of supporting time.

We will retain full support for legacy files forever.