whitequark/solvespace #54: Adding/removing point to existing spline

whitequark, 2016-01-11:
See http://solvespace.com/forum.pl?action=viewthread&parent=430
Also, it should probably be possible to have more than ten.

whitequark, 2016-04-10:
Fixed in 0939d851 and db4a3185c.

whitequark, 2016-04-12:
Okay, we need to do something with constraints. Currently if the points are used in any constraints they get really messed up.

whitequark, 2016-04-13:
@Evil-Spirit this doesn't seem to do anything.

whitequark, 2016-04-13:
Oh, ReplacePointInConstraint only handles point-coincident constraints... hrm.

whitequark, 2016-04-13:
@jwesthues Was there any particular reason for only touching point-coincident constraints in `ReplacePointInConstraints`?

jwesthues, 2016-04-13:
No special reason; I think I just didn't want to bother testing all the constraints to confirm that there were no surprising special cases.

whitequark, 2016-04-13:
Ah, OK. I'll remove the condition and test this, then (same for `FixConstraintsForPointBeingDeleted`).

jwesthues, 2016-04-13:
Correct behavior for `FixConstraintsForPointBeingDeleted` may be less obvious. I added that because if you constrain points {A,B} and {B,C} coincident, it's surprising if after you delete point B, A and C are no longer constrained. I'm not sure there's an obvious analogy to that for other constraints.

whitequark, 2016-04-13:
Right, I've just discovered that. The right behavior here is just to remove all non-point-coincident constraints related to the point being deleted, so I'll add another method on top of that.

jwesthues, 2016-04-13:
That already happens in `SolveSpace::PruneConstraints` upon regeneration; so the only thing you'd gain from that is to make the "x constraints deleted" message disappear.

whitequark, 2016-04-13:
The problem is that when we delete a point in the middle of a bezier, another one immediately takes its place (except in case of the first and last one), which causes the constraints to reapply to it, and distort the bezier in surprizing ways.

jwesthues, 2016-04-13:
That makes sense. I'd forgotten we were talking about the spline thing and not deletion in general.

whitequark, 2016-04-13:
Fixed in 9eaccea063 and 7f9b76cd5c9, now with constraints sticking to their respective points.

closed 2016-04-13 by whitequark