whitequark/solvespace #57: Lathe operation crash SolveSpace

Symbian9, 2016-01-12:
Using latest nightly build (2016-01-11) have a bug with "Lathe" operation.
Any lathe will crash program immediatly.

System: LinuxMint amd64

whitequark, 2016-01-12:
Cannot reproduce. Can you confirm that you are using this package? https://github.com/whitequark/solvespace/releases/download/nightly.2016-01-11.1/solvespace_2.1_amd64.deb

whitequark, 2016-01-12:
I've pushed a new nightly that includes a `solvespace-dbg` package: https://github.com/whitequark/solvespace/releases/tag/nightly.2016-01-12. Would you please install them both and `gdb`, then run `gdb --args solvespace`, and when it crashes, run `bt` in the gdb prompt?

Symbian9, 2016-01-12:
Ok, i will try latest fresh builds and try make debug it.

Symbian9, 2016-01-12:
@whitequark, so I made tests, and approve bug in latest builds (2016-01-12).
After reproducing bug and using your instructions I take next reports:

- in GDB 
<code>~ $ gdb --args solvespace
GNU gdb (GDB) 7.6.1-ubuntu
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /usr/bin/solvespace...Reading symbols from /usr/lib/debug/.build-id/63/f12a52936b9c208a06676aaef32aa015f09c59.debug...done.
done.
(gdb) bt
No stack.
</code>

- in Terminal
<code> ~ $ solvespace
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Segmentation fault
<code>

Also attach my simple rectangle that I test to Lathe (any other figures also crash program).

@whitequark, so I made tests, and approve bug in latest builds (2016-01-12).
After reproducing bug and using your instructions I take next reports:

- in GDB 
<code>~ $ gdb --args solvespace
GNU gdb (GDB) 7.6.1-ubuntu
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /usr/bin/solvespace...Reading symbols from /usr/lib/debug/.build-id/63/f12a52936b9c208a06676aaef32aa015f09c59.debug...done.
done.
(gdb) bt
No stack.
</code>

- in Terminal
<code> ~ $ solvespace
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Segmentation fault
<code>

Also attach my simple rectangle that I test to Lathe (any other figures also crash program).

whitequark, 2016-01-13:
@Symbian9, sorry, I gave you incorrect gdb instructions... it should be first `run`, then `bt`.

Symbian9, 2016-01-13:
<code>(gdb) run
Starting program: /usr/bin/solvespace 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeafce700 (LWP 22916)]
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4c1619c in std::string::assign(std::string const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff4c1619c in std::string::assign(std::string const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00000000004951ab in operator= (__str=..., this=<optimized out>)
    at /usr/include/c++/4.6/bits/basic_string.h:542
#2  operator= (this=0x3301c50)
    at /home/travis/build/whitequark/solvespace/src/sketch.h:304
#3  operator= (this=0x3301c50)
    at /home/travis/build/whitequark/solvespace/src/sketch.h:444
#4  SolveSpace::IdList<SolveSpace::Entity, SolveSpace::hEntity>::Add (
    this=0x894680 <SolveSpace::SK+64>, t=0x7ffffffe9100)
    at /home/travis/build/whitequark/solvespace/src/dsc.h:294
#5  0x0000000000491853 in SolveSpace::Group::MakeLatheCircles (
    this=0x7fffea1755c0, el=0x894680 <SolveSpace::SK+64>, 
    param=<optimized out>, in=..., pt=..., axis=..., ai=<optimized out>)
    at /home/travis/build/whitequark/solvespace/src/group.cpp:682
#6  0x00000000004934fe in SolveSpace::Group::Generate (this=0x7fffea1755c0, 
    entity=0x894680 <SolveSpace::SK+64>, param=0x894690 <SolveSpace::SK+80>)
    at /home/travis/build/whitequark/solvespace/src/group.cpp:427
#7  0x0000000000488402 in SolveSpace::SolveSpaceUI::GenerateAll (
    this=<optimized out>, first=2, last=2, andFindFree=false)
    at /home/travis/build/whitequark/solvespace/src/generate.cpp:239
#8  0x000000000049402c in SolveSpace::Group::MenuGroup (id=<optimized out>)
    at /home/travis/build/whitequark/solvespace/src/group.cpp:237
#9  0x00007ffff71230a0 in Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*)
    () from /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1
#10 0x00007ffff220b3b7 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff2223e82 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff2224ae2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff65f147c in gtk_widget_activate ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#14 0x00007ffff64f3edd in gtk_menu_shell_activate_item ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#15 0x00007ffff64f422c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#16 0x00007ffff64e2645 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#17 0x00007ffff220b188 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007ffff221c8db in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007ffff22244f9 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff2224ae2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff65f25f4 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#22 0x00007ffff64e0df4 in gtk_propagate_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#23 0x00007ffff64e11ab in gtk_main_do_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#24 0x00007ffff615c27c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#25 0x00007ffff1f423a6 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007ffff1f426f8 in ?? ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007ffff1f42afa in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff64e0277 in gtk_main ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#29 0x00007ffff712132d in Gtk::Main::run(Gtk::Window&) ()
   from /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1
#30 0x000000000042bb53 in main (argc=1, argv=0x7fffffffe0b8)
    at /home/travis/build/whitequark/solvespace/src/gtk/gtkmain.cpp:1556
(gdb)</code>

whitequark, 2016-01-13:
Fixed in https://github.com/whitequark/solvespace/releases/tag/nightly.2016-01-13.

closed 2016-01-13 by whitequark