whitequark/solvespace #60: "Length Ratio" between a line segment and an arc

Symbian9, 2016-01-19:
Please, add to presented "Length Ratio" constraint possibility to use line and arc as base.
For now I use next steps to reproduce needed result:
1) sketch arc
2) sketch line#1
3) constraine line#1 with arc using "Equal length"
4) sketch line#2
5) constrain line#2 with line#1 using "Length Ratio"

So, this future will simplify this task and remove 2 additional steps.

P.S.: Also, it will be good to add circle length to "Equal length" and "Length Ratio" constraints.
Thanks for cool CAD!

whitequark, 2016-01-26:
I'll consider this

whitequark, 2016-02-19:
Probably not. "Length difference" required far more construction so it was worthwhile, but here it's just one line; same applies to circle length for length equality and ratio constraints.

closed 2016-02-19 by whitequark