whitequark/solvespace #62: Free move background image

Symbian9, 2016-02-06:
Add possibility move imported image with mouse.
It give better usage when user have "3view" image of thing that want reproduce.
Maybe, its possible make import image in plane?
--
For now its difficult use background because need crop 3view to 3 images, setup point of import and other not userfriendly manipulations.

whitequark, 2016-02-06:
Hm, reasonable

Symbian9, 2016-02-11:
I like how background templates realised in PolygonCrest ( http://ysflight.in.coocan.jp/polygoncrest/e.html )
- for Windows (source+binary in zip)
http://download.cnet.com/Polygon-Crest/3000-6677_4-76170331.html
- for Linux, Mac (source+binary in zip)
http://download.cnet.com/Polygon-Crest/3000-6677_4-76169623.html
In PolygonCrest there is menu  "View --> Reference Bitmap --> Open Reference Bitmap Dialog"

whitequark, 2016-04-22:
@jwesthues What do you think about moving the "Background image" option to workplane properties? I.e. background image filesystem path and scale (and maybe add a setting for alpha) would become per-workplane; images would be rendered in the same plane as the workplanes, and if their workplane is visible.

Evil-Spirit, 2016-04-22:
@whitequark, may be even make it entity? to make ability to resize/move parametrically

Evil-Spirit, 2016-04-22:
We can make request image and introduce image frame (generated entities).
We can insert image and make some points aligned with image points and constrain it with image frame, after we can introduce image scale by appling known image distance.

[image]

whitequark, 2016-04-22:
It's complex and I don't see a compelling use case for this complexity.

Evil-Spirit, 2016-04-22:
I've just demonstrated the use-case. We load drawing(or photo) image and introduce scale for it using the solver.

jwesthues, 2016-04-22:
So how do people want to use this feature? I implemented it so that I could do ugly stuff like scaling off a bitmap drawing or photograph, but @Symbian9's use case seems different.

whitequark, 2016-04-22:
As I understand @Symbian9 wants to trace three 2d axis-orthogonal projections and reconstruct a 3d model. This seems to naturally call for assignment of projections to workplanes.

jwesthues, 2016-04-22:
So still for tracing only, so the import is still temporary (and not saved in the file)? In that case, I'd agree that your per-workplane option is preferable.

Evil-Spirit, 2016-04-22:
@jwesthues, Symbian9 will get ability to adjust image positions with solver. For example, he have some some image with circle hole. He can load the first image where hole is drawn as circle and do extrusion. After, he can load the second image with side hole view (with axis line). He just draw this axis line, link it to image and constrain to be on the extruded cricle axis. I actually don't see any complexity on this, for example in AutoCAD we can arbitrary number of images and arrange it like you want. May be user just wanted to insert his company logo as it drawn with photoshop.

Evil-Spirit, 2016-04-22:
@jwesthues, not so real example for positioning, but I believe we can found one reasonable.

Symbian9, 2016-04-23:
Yes, I want use 3 or more images for reconstruct only, and not for inbuilt inside SLVS-file.
So, as I sad good example is in opensource PolygonCrest (see previouse links)
Another one example is plugin for Blender:
* https://wiki.blender.org/index.php/Extensions:2.6/Py/Scripts/Add_Mesh/Planes_from_Images
* https://youtube.com/watch?v=wu7fW4UYFlw

I agree with idea for temporary importing image as sketch in plane.
Also, as for me, it would be good if user can open few images at same time, that will better understanding how model will look
[image]
* https://forum.flightgear.org/viewtopic.php?p=207837

Evil-Spirit, 2016-05-04:
@whitequark, can I start implement this?
I can use any approach, but I insist on the implementing the image-request.
It's not so harder in implementation, but more flexible.
We will get ability:
1. Move, rotate, scale image as we need.
2. Use solver to choose position, scale, aspect ratio, angle.
3. Insert any number of images in any planes (or just inside 3d group).

Symbian9, 2016-08-23:
@Evil-Spirit, any news?

Evil-Spirit, 2016-08-23:
@Symbian9, not priority for now.

whitequark, 2016-11-27:
@Evil-Spirit You've convinced me, let's go ahead with image request.

whitequark, 2016-11-27:
@Evil-Spirit For now let's implement it like this:

  * Four points in corners, like in text request.
  * Move, rotate, scale transformations.
  * If "bool other" enabled then allow any affine transform (independently draggable points).
  * No need to do cropping for now, I think. We aren't a general purpose image processing software, we don't need crop, changing colors, etc. This all can be more easily done with gimp, we should develop one tool and do it well.
  * Same logic for loading when picture is missing as for linked files. Show some "X" when the picture is not found.

Evil-Spirit, 2016-11-29:
@whitequark 
https://github.com/Evil-Spirit/solvespace/commits/master-image
WIP

Evil-Spirit, 2016-11-30:
@whitequark,
almost finished here https://github.com/Evil-Spirit/solvespace/commit/b40f3a71a84d682b79e0b0bbc5242798b730b3e2

ready:
* Four points in corners, like in text request.
* Move, rotate, scale transformations.
* Same logic for loading when picture is missing as for linked files. Show some "X" when the picture is not found.

not ready:
* If "bool other" enabled then allow any affine transform (independently draggable points).

because it needs to change canvas interface to have ability draw some arbitaraty quad-pixmap.

Evil-Spirit, 2016-11-30:
[image]
[image]

whitequark, 2016-12-01:
> If "bool other" enabled then allow any affine transform (independently draggable points).

Let's skip this until someone requests it, it's not important.

Evil-Spirit, 2016-12-01:
@whitequark,
So, you can review now, may be I forgot something.

Symbian9, 2016-12-01:
> Let's skip this until someone requests it, it's not important.

Ok, I request it now!

Evil-Spirit, 2016-12-02:
@Symbian9, What is your case to use such transform?

whitequark, 2016-12-05:
@Evil-Spirit I see no obvious problems in review. The use of the "font" field is kind of gross but we'll deal with that during migration to the new savefile format anyway, like with "other".

Evil-Spirit, 2016-12-05:
@whitequark, for not confusing, we can make methods like ImageGetRelFileName, ImageGetFileName which will return str and font

whitequark, 2016-12-22:
@Evil-Spirit changes:
  * No more absolute path stored in savefile. Absolute path is only for compatibility with SolveSpace 1.9.
  * Simplified code for adding request, reusing LoadLinkedImage.
  * Linked images are never reloaded during drawing, only during generation.
  * Regenerate All (space) reloads images, so that any updated images are redrawn.
  * Use-after-free in rendergl2 fixed.

In the future do **not** use absolute paths anywhere, they make behavior worse. E.g. imagine a.slvs linked to b.slvs in same folder. Now copy the folder somewhere else. Instead of a proper missing file dialog you will get a spurious file linked, and the sketch will break altogether if sent to another computer.

This is why MakePathAbsolute/Relative was private to file.cpp.

whitequark, 2016-12-22:
Actually, there are still some issues with absolute paths (when saveFile=="") which I will resolve later...

Evil-Spirit, 2016-12-23:
@whitequark, where I can look at this?

whitequark, 2016-12-23:
Branch image in this repo

Evil-Spirit, 2016-12-23:
> Instead of a proper missing file dialog you will get a spurious file linked, and the sketch will break altogether if sent to another computer.

Hm. We are looking at relative path first, then absolute, so we can copy folder with project anywhere we want.

whitequark, 2016-12-23:
> Hm. We are looking at relative path first, then absolute, so we can copy folder with project anywhere we want.

And this does the wrong thing if you copy then delete a linked file.

Symbian9, 2016-12-23:
> * No more absolute path stored in savefile. Absolute path is only for compatibility with SolveSpace 1.9.

May I ask you, @whitequark, what is <b>SolveSpace 1.9</b>? 

As I know, only SolveSpace 2.0 and above availiable for downloading at this moment.

whitequark, 2017-03-12:
Merged.

closed 2017-03-12 by whitequark