whitequark/solvespace #67: Builtin vector font needs grid-fitting

whitequark, 2016-03-05:
[image]

Evil-Spirit, 2016-03-29:
@whitequark, can you provide more information about this issue?

whitequark, 2016-03-29:
You need to ensure that, in screen pixel coordinates, every point in the vector font has integer coordinates when the font size is less than 12pt. Just round it to nearest integer.

Evil-Spirit, 2016-03-29:
I don't really think what this make it better, just bring more problems. We can't break subpixel accuracy because we have arbitrary dynamic projection, so performing it will cause much more visual artifacts.

whitequark, 2016-03-29:
Of course it will, grid fitting is a standard technique.

whitequark, 2016-03-29:
I actually implemented a simple version of this, but it broke because it did not consider `SS.GW.scale`, the font rendering code is pretty annoying there.

Evil-Spirit, 2016-03-29:
It will cause visual flickering of text while the panning and rotation of view, i think it is the same kind of evil. The only point to apply it - while the exporting image.

whitequark, 2016-03-29:
Why would it? Do you see any flickering while a browser is scrolling? I don't understand; simply panning text in increments of one pixel is enough.

whitequark, 2016-03-29:
Basically, try it yourself: implement a hardcoded version that works only at one scale and you'll see that it looks just fine while panning.

Evil-Spirit, 2016-03-29:
OK, i will try

Evil-Spirit, 2016-04-05:
Without fitting
[image]

With fitting (looks ugly)
[image]

Evil-Spirit, 2016-04-05:
Actually it needs more powerful font-hinting algorithms.

ruevs, 2016-04-05:
"With fitting (looks ugly)" actually looks much better to me.

whitequark, 2016-04-05:
@Evil-Spirit It looks worse on larger font sizes. That's completely expected, really. But it should look better on 12pt and below, where the more 'blocky' nature of the letters is nowhere to be seen.

Evil-Spirit, 2016-04-06:
without fitting
[image]
each vertex of font
[image]
glyph bounds
[image]

Which is better?

whitequark, 2016-04-06:
Hmmm. What if we grid fit only vertical/horizontal strokes?

Evil-Spirit, 2016-04-06:
I've tried it, but it needs real curves instead of approximating segments. The true type font-hinting is always hand-work(https://www.typotheque.com/articles/hinting, http://alistapart.com/column/font-hinting-and-the-future-of-responsive-typography). The more i can do - grid-fit lower-case level of glyphs (now only fit common bound of captial letters - zero level, height level and horizontal bounds(min max))

Evil-Spirit, 2016-04-06:
I think the last image is better.

ruevs, 2016-04-06:
The last one ("glyph bounds") looks best to me as well. The smallest size is almost readable and the larger ones look the best - both in Latin and Cyrillic. I made a side-by-side image to compare easier.
[image]

"ех, чужд кайф: сплющь..." nice one :-)

whitequark, 2016-04-06:
Can we do real curves? The source data, I.e. lff files, specify arcs explicitly, it makes sense to convert them to pwl in the font renderer.

Evil-Spirit, 2016-04-07:
The main problem is make hand work, with automatic hinting we can't get the best quality

whitequark, 2016-04-07:
Freetype's autohinter works quite well, and I don't see why we can't do it as well.

Evil-Spirit, 2016-04-07:
The llf uses arcs, we have to convert them to bezier anyway (we need to arrange derivatives too). I don't see why we need it now, the font looking much better, i think it's enough. In other hand, if want to be best in font rendering, we can use free type renderer.

Evil-Spirit, 2016-04-07:
Actually we also need to align width of each glyph (bounds) and interval between glyphs (to ensure each character has the same width at each string position.). I think it's not the problem, when we need to do 2d rendering (when text nerver goes 3d), when we need both (like comment in plane) we can't do it, because of ugly jittering.

Evil-Spirit, 2016-04-07:
[image]

Evil-Spirit, 2016-04-07:
http://www.paratype.com/services/truetypehinting.pdf

whitequark, 2016-04-07:
When text goes 3d then just don't do any hinting, it's OK

Evil-Spirit, 2016-04-07:
With aligning of widths and intervals, It will jump dramatically (can become 1.5 - 2 times shorter)

whitequark, 2016-04-07:
Yes. That's the cost of doing business, so to say.

Evil-Spirit, 2016-04-07:
If we want commented table-stamps for good appearance and printing, we just want to have same-string-lengths in view port and after the export. The EPS and PDF are drirectly-printing formats(executes inside printers and plotters), so we have to ensure the same appearance.

whitequark, 2016-04-07:
Well, this is problematic... PDF export doesn't really need hinting, since printers with less than 300dpi don't exist and as such hinting is pointless. Note that this problem remains even if we use Freetype.

Evil-Spirit, 2016-04-07:
Yes, this problem will be the same. So, i think, solvespace is not about observing details inside viewport so much, it's about designing and output for drawings (dxf, pdf, printer) or manufacturing (step, stl).

whitequark, 2016-04-07:
It's both. You can even do manufacturing with a CAD-only workflow, which at no point involves drawings (but SolveSpace is nowhere near supporting that yet).

Evil-Spirit, 2016-04-14:
@whitequark , don't forget what I insist on https://github.com/Evil-Spirit/solvespace/commit/a588a02b512db1a7659a42ef9e4a84c08c857475

Evil-Spirit, 2016-04-14:
@whitequark, and we can use this approach to make text constants(like capital, letter, overall height) https://github.com/whitequark/solvespace/compare/for-upstream...Evil-Spirit:font-grid-67#diff-6fc5e64f1f4fa1e5646c890ced044a17R334

whitequark, 2016-04-14:
Yeah, I remember about that one. How much larger does it make the compiled binary?

Evil-Spirit, 2016-04-14:
Not so large. I actully dont remember, but i checked it

whitequark, 2016-04-16:
@Evil-Spirit I've just tested your solution and results in extremely high quality rendering, on a wide range of font sizes starting from 6px. This is much better than what I expected. Thanks!

whitequark, 2016-04-16:
Fixed in e74ccb301071. And I also made lff2c automatically compute the rest of the font metrics.

closed 2016-04-16 by whitequark