whitequark/solvespace #68: Automatic constraints can cause issues

marcbritten, 2016-03-06:
Creating new geometry where the segments terminate on a line or point causes constraint issues when the program also generates an automatic H or V constraint.

I would recommend dropping the automatic constraint if both points of a segment are constrained already.

whitequark, 2016-03-06:
Can you show an example sketch? Constraining a segment horizontally or vertically makes sense even if both ends are constrained, e.g. try drawing a rectangle.

marcbritten, 2016-03-06:
Create a rectangle, then extrude it.  Now place a line segment between two points in a way that creates a horizontal constraint as well.

whitequark, 2016-03-07:
Yes, I see the problem you're talking about. However, I maintain that adding an automatic constraint is useful even when both ends of a segment end up constrained to something else. Can we solve this problem in some other way? E.g. `Ctrl` turning off the autoconstrainter temporarily?

whitequark, 2016-03-15:
Ok, yeah, I've started hitting this problem a lot. I think I'll turn it off if `Ctrl` is pressed, as a solution.

Evil-Spirit, 2016-03-29:
The real solution should be based on dof analyzing, but it might be problematic. Should I introduce ctrl key?

whitequark, 2016-03-29:
Yes.

whitequark, 2016-04-08:
Fixed in 3d334d1.

closed 2016-04-08 by whitequark