whitequark/solvespace #72: Could NOT find SPACEWARE

avh-on1, 2016-03-10:
Building from git repo (6e56b00b9ae30b31df677cbe8fefdfef7edb7ad7) on Fedora 23, cmake returns this error:

`-- Could NOT find SPACEWARE (missing:  SPACEWARE_LIBRARY SPACEWARE_INCLUDE_DIR) `

whitequark, 2016-03-10:
Do you want to use a SpaceWare input device?

avh-on1, 2016-03-10:
Nope

whitequark, 2016-03-10:
It's not a fatal error and won't prevent you from building SolveSpace.

closed 2016-03-10 by whitequark

avh-on1, 2016-03-10:
Also, I do have libspnav-devel

whitequark, 2016-03-10:
This is what I have in `FindSpaceWare.cmake`:

```
    find_path(SPACEWARE_INCLUDE_DIR
        spnav.h)

    find_library(SPACEWARE_LIBRARY
        NAMES spnav libspnav)
```

Are these header and library in your system include and library paths?

avh-on1, 2016-03-10:
I have

    /usr/lib64libspnav.so
    /usr/include/spnav.h
    /usr/lib64/libspnav.so

whitequark, 2016-03-10:
Can you pastebin the content of `CMakeFiles/CMakeOutput.log` in the build directory?

avh-on1, 2016-03-10:
[here you go](http://pastebin.com/x2wtmpDX)

whitequark, 2016-03-10:
Also `CMakeError.log`?

avh-on1, 2016-03-10:
Rebooted my computer overnight.
`-- Found SPACEWARE: /usr/lib64/libspnav.so`
Sigh... thanks for the prompt assistance.