whitequark/solvespace #73: "New Group -> Frustum"

Symbian9, 2016-03-15:
Can you add some option to "Extrusion" operation or make new "Frustum" operation for new group creation to realize "frustum"?
https://en.wikipedia.org/wiki/Frustum

Evil-Spirit, 2016-05-04:
@whitequark,
@jwesthues.

I have idea about this: We can introduce new kind of useful operation - "Loft Extrusion".
1. We need to create some number of sketches, shifed along extrusion vector.
2. We consider them as step and repeat in sense of disabled ability to remove/add entities (and change topology by removing/adding PC-constraints).
3. We make separate set of parameters for them and make ability to introduce new constraints( excluding p.2) for make alility to change each sketch separately.
4. We can easily create NURBS body for it using the current functionality.

The useful usecases:
1. Frustum groups (something like, but not 100% of cases)
2. Airplane wings with complex profile.

Evil-Spirit, 2016-12-07:
@whitequark, I've learn some book about NURBS, booleans and so on. For creating true frustum he fistly have to do equidistant for sketch (which exist in SketchFlat).

whitequark, 2016-12-07:
@Evil-Spirit what do you mean by equidistant? don't we have that already?

Evil-Spirit, 2016-12-07:
@whitequark 
I mean "offset" http://cq.cx/sketchflat.pl

whitequark, 2016-12-07:
@Evil-Spirit why? cannot we just scale the sketch?

Evil-Spirit, 2016-12-07:
@whitequark, It was my first thought, but this is geometrically wrong.

whitequark, 2016-12-07:
@Evil-Spirit why? can you show an example?

Evil-Spirit, 2016-12-07:
[image]

Evil-Spirit, 2016-12-07:
@whitequark,
Imagine bottom sketch as scaled. This is not the same.

Evil-Spirit, 2016-12-07:
I made sketch offset and top looks like exactly the same
[image]

Evil-Spirit, 2016-12-07:
@whitequark 
I am reading this book https://www.amazon.com/Geometric-Modeling-mathematics-Nikolay-Golovanov/dp/1497473195

whitequark, 2016-12-07:
@Evil-Spirit I thought a frustum is a section of a pyramid. But the solid above wouldn't become a pyramid if you shrank the top sketch towards the limit of 0 in area, no?

whitequark, 2016-12-07:
@Evil-Spirit Anyway. To support offset groups we'll need to derive entities, but we have degenerate cases where there are fewer entities in the output than in input. What do we do in this case?

Evil-Spirit, 2016-12-07:
@whitequark I think we can just generate zero-length entities and skip them while processing of result sketch

whitequark, 2017-01-02:
I've decided that adding new solid operations is less of a priority than migrating to the new file format and hierarchical sketches. At some point we will probably add the "loft extrusion" operation @Evil-Spirit is suggesting, but not very soon.

closed 2017-01-02 by whitequark