whitequark/solvespace #74: Coordinate system configuration

Symbian9, 2016-03-15:
From start SolveSpace show Y axis as vertical, but many other application from start use Z axis as vertical.
Think it would be good if SolveSpace will have configuration option for starting coordinate system.
Also, there are some difference between differente software, that can have some other axis coordination, so it will help to export in custom axis rotation.

whitequark, 2016-04-16:
@jwesthues What's your opinion on this feature?

jwesthues, 2016-04-16:
You already can export with a transformed coordinate system by a slightly clumsy method (create a single-part assembly containing the part with the desired transform). So there may be some value in a special feature for that, but not a lot.

SolveSpace has no particular notion of "up", beyond the initial default view and sketching plane. The standard argument for z-up is that a plan view is "most important" (in architecture, etc.), and should therefore be the xy plane. So even then, it still seems reasonable for the default sketching plane to be xy.

closed 2016-04-16 by whitequark