whitequark/solvespace #75: "Paste mirrored" (2D sketch)

Symbian9, 2016-03-15:
Sometime when you near complete detailed drawing and constrained it you apologize that it must be... mirrored. So, for solve this problem it will be good make option to "Copy -> Paste mirrored" with choosing axis

whitequark, 2016-03-15:
Edit → Paste Transformed... can do it.

closed 2016-03-15 by whitequark

Symbian9, 2016-03-15:
This is not intuitive and it have NOT "mirror" future.

whitequark, 2016-03-15:
Set scale to -1.

ryannining, 2016-07-04:
i have issue too, while we can use new group rotate by a point and line to do mirror. But the resulted sketch have inverted normal axis so it cannot be extruded.

I think a new Mirror (new group or paste) will do. I wish i can do coding to help you guys.

whitequark, 2016-07-04:
@ryannining Sure it can be extruded. See the following file for an example of how to do it:
[mirror.zip](https://github.com/whitequark/solvespace/files/346753/mirror.zip)

ryannining, 2016-07-05:
thanks, it work

whitequark, 2019-05-24:
https://github.com/solvespace/solvespace/issues/227