whitequark/solvespace #77: Paste Transformed on right mouse button

Symbian9, 2016-03-15:
For now when something copied and than click right mouse button there is only "Paste" tab.
Please, add in this contextutal menu one more tab with "Paste Transformed" option.

whitequark, 2016-03-15:
I don't think there is any point in this, given that even "Paste" doesn't take mouse position into account (maybe it should); for "Paste Transformed" the position is fully determined by the settings you enter in the text window, so the context menu option doesn't add anything of value.

closed 2016-03-15 by whitequark

Symbian9, 2016-03-15:
But right button tab is more ergonomic and quicker than "Ctrl+T" or "Menu -> Edit -> Paste Transformed..." way.
In SolveSpace use right mouse button must be preferable, and short keys and menu tabs are additional.

whitequark, 2016-03-15:
What? You can't do almost anything through RMB.

Symbian9, 2016-03-15:
For now RMB already have many option:
- Assign to Style
- Group Info
- Style Info
- Select Edge Chain
- Cut
- Copy
- Paste
- Delete
- Unselect All
- Unselect Hovered
So, as for me as for "mouse user" right button menu and icon tabs menu is preferable than short key assigns.

whitequark, 2016-03-15:
Yes. This is a very small subset of operations, most of which have something to do with the entity under cursor.

Symbian9, 2016-03-15:
So, what a problem add "Paste Transformed" to this list?

whitequark, 2016-03-15:
e87e787.