whitequark/solvespace #80: Pie menu for SolveSpace (ideas)

Symbian9, 2016-03-15:
For now I try create concept of "pie menu" specially for SolveSpace.
https://en.wikipedia.org/wiki/Pie_menu
There is a online tool that can be used for concept design of "pie menu"
http://pmg.softwaretailoring.net/
Target is better ergonomic for "mouse user" and "touchscreen user".
1) https://doc.qt.io/archives/qq/qq11-piemenu.html
2) https://wiki.blender.org/index.php/Extensions:2.6/Py/Scripts/3D_interaction/Pie_Menu
3) https://wiki.blender.org/index.php/Dev:Ref/Release_Notes/2.72/UI/Pie_Menus
4) http://forum.freecadweb.org/viewtopic.php?t=10892
5) http://www.openorienteering.org/news/2012/openorienteering-mapper-0-4-provides-a-wealth-of-new-features/

whitequark, 2016-03-15:
Please ask Jonathan first.

Symbian9, 2016-03-15:
Discussion HERE:
http://solvespace.com/forum.pl?action=viewthread&parent=1236

whitequark, 2016-03-18:
Reopen once you get feedback from Jonathan.

closed 2016-03-18 by whitequark

Evil-Spirit, 2016-04-15:
It seems we have no any intention to build under Android or IOS. This is a lot of work to reconsider all the user interaction for touch-screens, there is not only about implementation of the pie menu. Firstly, we could refactor the code to separate user-interaction code from the logic of sketch-editing.

whitequark, 2016-04-15:
Yeah, that's a prerequisite in any case.

whitequark, 2016-04-15:
@Evil-Spirit By the way, I do have an intention for at least an Android port. There is no even half-decent CAD that works on Android and there is a very large amount of people that only have Android tablets. I think this is a massive, unexplored niche which SolveSpace could fill with relatively little difficulty. Traditional CADs aren't going to do that any time soon, in no small part because they have so much legacy code and concepts tied to PCs...

whitequark, 2016-04-15:
And the existing SolveSpace UI actually maps quite well to tablets already. Swipe from the right side to get access to the text window, swipe from the left to access the menu, replace toolbar with a pie menu. So we don't even need any large changes.

I might give it a try when I have some free time...