whitequark/solvespace #85: Wrong constraint set is exported to DXF

whitequark, 2016-03-16:
Constraints in non-last groups are exported.

Evil-Spirit, 2016-03-22:
I think it is need to make a checkbox for showing/exporting constraints for choosen groups like it is performed now for an entities.

whitequark, 2016-03-23:
No. At some point there should be a separate drawing module. Before that let's keep it simple.

Symbian9, 2016-03-23:
I also think about this future. As for me, it will be good if constraints will have two additional inbuild styles:
 1) s00e-#def-constraint-printed
 2) s00f-#def-constraint-not-printed

Think it would be simpler than make additional checkboxes.
Also it will add possibility customise different constraints.

Another way is add "Toggle construction" option applied to constraints too.

whitequark, 2016-04-08:
Fixed in e17a2481.

closed 2016-04-08 by whitequark