whitequark/solvespace #91: Show unit symbols in constraints

Symbian9, 2016-03-23:
Examples how it can look: 
10.4″ - length in inch
264.16mm - length in millimetrs
⌀10 - diametre (also "⌀10mm")
R10 - radius (also "R10mm")
45° - angle size (and maybe like this "∠45°")

Think its possible to use code that presented for referenced constraint marking (show or hide  "REF" after numbers):
100.5 - not referenced constraint  
100.5 REF - referenced constraint

So, if it would be implemented, then it will look like "100mm REF" or "R12 REF", etc.

whitequark, 2016-03-23:
No. Using different units on the same drawing is insane, and if you're not, you don't need them in labels. R and ⌀ symbols are already shown, what's the problem here?

closed 2016-03-23 by whitequark

whitequark, 2016-03-23:
I added the degree symbol to angle constraints though.

Symbian9, 2016-03-23:
I'm NOT tell use different units in same drawing!
Only show current global unit after dimension. Sorry, about R and . Yes, its shown.
But anyway, angles need degrees symbol.
Read this - http://www.pages.drexel.edu/~rcc34/Files/Teaching/MEM201%20L5-Fa0809-SpDimensions_RC.pdf
 Also this document describe that referenced dimension must be shown as "(R10)" instead then "R10 REF".

Think need deeply view to technical drawing standarts, widely used in the world:
1) https://en.wikipedia.org/wiki/ISO_128
2) https://en.wikipedia.org/wiki/GOST
3) http://cadsetterout.com/technical-drawing-standards/

whitequark, 2016-03-24:
> Only show current global unit after dimension.

No. The entire drawing uses the same symbol. Displaying it everywhere is just visual noise.

> anyway, angles need degrees symbol.

I implemented those.