whitequark/solvespace #92: Make bigger text input field

Symbian9, 2016-03-23:
For now text input field  look too small (I mean font size in input field). So, if its possible, please, enlarge it to double size or so.

whitequark, 2016-03-23:
In the text window, no can do; it's bound to the font size in the text window.

In graphics window, yes, it should be bound to the text size of the constraint style.

Evil-Spirit, 2016-03-25:
Do you have ultra hd display and have you enlarged default font size?

Symbian9, 2016-03-25:
@Evil-Spirit I have 1280*800 in 15", but text inside input field for constraints look small.

Evil-Spirit, 2016-03-25:
@whitequark i think it appears only on Linux. Can we create os-specified labels for issues (windows, linux, macos, etc.)?

whitequark, 2016-03-25:
What appears only on Linux?

You need to make the text field size match the constraint style. This is an issue on every platform. You only need to handle the Windows port though, I'll handle the rest.

Symbian9, 2016-04-09:
In latest nightly input field bigger, but there is another issue - input field depen from position of comment element, in some situation it can disappoint user where it would gone.
Some next step is make input field always centere in main window.

Evil-Spirit, 2016-04-10:
This changeset is not yet integrated.

whitequark, 2016-04-11:
@Evil-Spirit The font size is wrong. For the default constraint size, the font size should match the old font size; they corresponded to each other.
Old: 
[image]
New: 
[image]

Evil-Spirit, 2016-04-12:
The comment reworked, the problem was because of too large minimal editor text height.
But anyway, i have to rework some magical-fitting (which is broke when i adjust min height).
This (quite simple) problem takes too much time; i suggest the future things with platfom-dependence (winapi is hell) implement in opengl (platform-independed). The time to implement this is comparable, but the result will be much-better-looks-like (for ex. look at my unfinished gui library https://github.com/Evil-Spirit/Nutmeg/blob/master/CoreSrc/Gui/GuiEdit.h).

whitequark, 2016-04-12:
No, we will only use native edit controls. For example, your fancy OpenGL edit controls will not support input methods, required for Asian characters.

whitequark, 2016-04-12:
Fixed in a4c35f490.

Evil-Spirit, 2016-04-13:
Please, see what i do with my commits; your implementation don't actually fit the text
[image]
[image]

Evil-Spirit, 2016-04-13:
Also, the font size was broken. The AutoCAD assumes it like capital height. GOST also says about height of the capital charactes(http://www.pntd.ru/2.304.htm)

Evil-Spirit, 2016-04-13:
[image]
[image]

whitequark, 2016-04-13:
Re: fitting the text width or using a minimum text size. I did that deliberately. It's a distinct change, doing it in platform-dependent code is the wrong place to do it, and it introduces its own problems (like editing a comment named just "l" will result in a text field too short). We can add that separately, but it's out of scope for this issue.

Re: capital height: this is a more complex issue than I thought. Apparently there are two terms:
  * Font _size_, which is defined as baseline distance or, more commonly in computer graphics, ascent + descent.
  * Font _height_, which is defined as height of a capital letter.
(from [DIN 16507](https://www.cl.cam.ac.uk/~mgk25/metric-typo/))

In the code I was erroneously using `fontHeight` where I really meant `fontSize`. That needs to be fixed. In any case, it would be prudent to use `ssglCapHeight` during export to DXF and so on; we do not have to change our implementation to match what DXF does.

whitequark, 2016-04-13:
Fixed font metrics to correctly resemble the input in the UI and output to DXF in e2c7dad25 and b6ddebf3d.

Re: edit control width, an acceptable solution would be something like...
  * Query the OS font rasterizer for the width of the text;
  * Ensure the field is not too short;
  * Trim the field against the right-side border (or left-side border in case of RTL text)...

but it's a lot of work and I don't think it's very important. The current solution works fine.

Evil-Spirit, 2016-04-14:
This looks ugly, anyway, this is time to do it; We should not bother about too small or too big text, we just have to overlap screen-space taken by string, this calculations based on vector-font-metrics. Os string may be smaller or bigger (which is actually happens now).

whitequark, 2016-04-14:
> have to overlap screen-space taken by string, this calculations based on vector-font-metrics.

No, I don't think this makes any sense. First, the vertical dimension of the edit control is calculated based on the height of the system font, not the vector font. Second, what exactly is the point of overlapping the text in the vector font? It doesn't improve the UI in any way.

Evil-Spirit, 2016-04-14:
Just try to use it, it is better when you have some logic, not the constant size evert time.  90percent of cases is numbers which is fit. Iiiiiiiiiiiiiiiiiiiiiiiiii is the not real case

whitequark, 2016-04-15:
Yep, I tried to use it. I think it's a very minor improvement and I don't want to deal with implementation complexity right now, we have more important stuff to do.

Evil-Spirit, 2016-04-15:
@whitequark, the point is what i've already done it, but you just cancel this changes, but it have a lot of time wasting (in the future is never) to not inculde it.

whitequark, 2016-04-15:
No, you didn't do it. You did it only for Windows and you put platform-independent logic into Windows-specific code.

Evil-Spirit, 2016-04-15:
We need just to extract it with no problems.

whitequark, 2016-04-15:
Ok, we can implement it. A field width of, say, 5 average width characters for numeric constraints and 30 average width characters for comments, or whatever's necessary to fit the contents.

Evil-Spirit, 2016-04-15:
OK.

whitequark, 2016-04-16:
Done in 1622927ac3.

closed 2016-04-16 by whitequark