whitequark/solvespace #95: Draw extension lines for angle constraints between line segments

whitequark, 2016-03-24:
Pretty confusing now.

Evil-Spirit, 2016-04-09:
I hate to do such things without operators overloading.
Why we should use ugly code instead of clean and simple one?
Simple to write, simple to maintain, simple to understand by anyone.

whitequark, 2016-04-09:
The upstream doesn't want operator overloading. So there won't be any.

Evil-Spirit, 2016-04-09:
[image]

Evil-Spirit, 2016-04-09:
What about appearance like this?

whitequark, 2016-04-09:
ISO GPS should say something about extension lines...

Evil-Spirit, 2016-04-09:
This is artificial restriction not based on the reality, we should change his mind.

whitequark, 2016-04-09:
I don't really see any difference between the styles myself, and it's a lot of pointless churn, so I don't think it's a good idea either.

Evil-Spirit, 2016-04-09:
Math can be simple at moment while you are in a context, when you lost it, all the + and->Minus() becomes obvious.

Evil-Spirit, 2016-04-09:
Call for @jwesthues

jwesthues, 2016-04-09:
Many projects use operator overloading, but many other projects don't. Some popular languages don't support it at all. We could find examples of serious mathematical software in almost any conceivable style. I weakly like that the current style immediately distinguishes between operations on vector or symbolic types and operations on numerical scalar types, and strongly dislike churn.

Are you sure you've done everything possible to make the current style understandable to you? (Indent appropriately? Break long expressions into smaller ones when necessary? Use an editor that matches parentheses well? Do your math on the whiteboard first instead of in the code?)

Returning from that aside, the feature looks nice. Is the arrowhead at 180 degrees a bug, though?

Evil-Spirit, 2016-04-09:
Some languages (like Java) has no operator overloading. For many people it is the reason why they don't use it for 3d application development. 
Vector algebra introduces vector operations and sybols for it, so why don't use it? why you need to see the vector and the scalars as different operations, this is just the same in abstraction.
There is no any charn. I think every graphics programmer use it while the creating shaders, and there is no any problems to understand where is the vector and where is the scalar. Expressions  are simple and intuitive, looks like everybody except.

  I am sure, while the using operators overloading it is no reason to indent, break long lines, match paretheses, break into small parts, etc. It' just the expression itself. Just math what we used to write, what we used to read and understand.

Yep, 180deg some buggy., Thanks god this is not yet ->plus->minus corrupted and i can found some reason for it :)

whitequark, 2016-04-09:
The reason people don't use Java for graphics programming is the lack of value types, leading to massive GC churn on trivial operations, not lack of operator overloading.

Churn is the very large amount of code change required to change the style. You have to rewrite pretty much all math.

Evil-Spirit, 2016-04-09:
And this too.
I've done it once, there is no reason why i can't do it again.

whitequark, 2016-04-10:
The question is not whether you can, but whether you should. Rewriting code breaks git history and introduces bugs.

whitequark, 2016-04-10:
@Evil-Spirit Extension lines look nice indeed, let's go ahead with this.

Evil-Spirit, 2016-04-11:
[image]

whitequark, 2016-04-11:
Excellent!

whitequark, 2016-04-11:
Problem 1: extremely large extension lines with small angles like this:
[image]
Problem 2: moving the label around with mouse is very unstable in this case:
[image]
Problem 3: text intersecting with the lines:
[image]

whitequark, 2016-04-11:
These problems aren't introduced *with* the extension lines patch but we should address them anyway.

Evil-Spirit, 2016-04-11:
I also suggest to use %g format specifier to avoid pointless ".00"

Evil-Spirit, 2016-04-11:
1. What we should do in this case?

whitequark, 2016-04-11:
Hm, good question. I actually don't know. Librecad has the same problem...

whitequark, 2016-04-11:
Looks like even AutoCAD doesn't have a good answer to 1)

whitequark, 2016-04-12:
@Evil-Spirit Ah I finally realized what's wrong here. We can solve problems 2 and 3 by making the label for the angle constraint interact with the line in the same way as for the distance constraints:
[image]

Evil-Spirit, 2016-04-13:
@whitequark, There will be no problems with image you have provided, but for small arcs it will be very bad appearance. OffsetCalculation (using EllipticalInterpolation) is performed in wrong way.
We should do something like this http://pyx.sourceforge.net/manual/box.html

whitequark, 2016-04-13:
@Evil-Spirit Yes. We already have that problem with distance constraints though:
[image]

I think what we should do is use the current format (dimension in middle of a line) for both angle and distance constraints, and, separately, detect when the dimension line is too small and take the label out of it. Then we don't even need your fancy algorithm, because you can assume that the line is nearly straight.

Evil-Spirit, 2016-04-13:
[image]

Evil-Spirit, 2016-04-13:
Problem 1:Don't know how to fix
Problem 2:Wontfix; just fix it in next iteration when we will introduce constraint offset basis.
Problem 3:Solved

whitequark, 2016-04-13:
Great!

whitequark, 2016-04-13:
@Evil-Spirit Can you also make the arrows invert and the line extend, like with distance constraints?
[image]

Evil-Spirit, 2016-04-14:
@whitequark OK, forget about it.

whitequark, 2016-04-15:
@Evil-Spirit can you please finish this first? I'd like to finish this merge and make a nightly.

Evil-Spirit, 2016-04-15:
Ok, will do it now.

Evil-Spirit, 2016-04-15:
[image]

whitequark, 2016-04-15:
@Evil-Spirit Good but that doesn't include the arrow inversion (like on distance constraints).

Evil-Spirit, 2016-04-16:
@whitequark, i don't know, but this can cause confusion. I dont found any samples about it.

Evil-Spirit, 2016-04-16:
@whitequark When we have linear dimension - we have straight line which is not closed-loop. And we can treat it correctly. When we have angular dimension, we have an arc which is loops in circle, and there is ambigous: we can treat it an aplha angle or 360-alpha.

whitequark, 2016-04-16:
@Evil-Spirit This is specified in ISO--e.g. see page 8 and especially page 38 of http://engineering.pages.tcnj.edu/files/2012/02/dimensioning_and_tolerancing.pdf. We really need to get a copy of ISO 128-22 somewhere...

Evil-Spirit, 2016-04-16:
@whitequark, I saw this document, but found only 8th page. Page 8th case can be applicable to the near 0 or 180 degrees cases. For page 38 we shoud consider what degree value is inside, but we have outside case and I have no opinion what we should do with.

whitequark, 2016-04-16:
@Evil-Spirit The direction of the arrow does not say anything about the angle--it's a part of the leader line from the label to the dimension being specified. I.e. think of the indicated angle as one separate thing and a label with a leader and an arrow pointing from it as another thing.

There isn't any ambiguity because angles over 180° aren't conventionally used.

See also https://ia801408.us.archive.org/18/items/BasicEngineeringDrawing/RhodesCook-BasicEngineeringDrawing.pdf for future reference starting at page 70--though it doesn't say anything new on this specific issue.

whitequark, 2016-04-16:
Actually, even if angles over 180° would be used there wouldn't be any real ambiguity because it is quite obvious when an angle is larger than half a circle.

Evil-Spirit, 2016-04-16:
> There isn't any ambiguity because angles over 180° aren't conventionally used.

Not used YET. We have to introduce it in future (for measuring arc angle which is actually 0-360).

whitequark, 2016-04-16:
Do we need that? It's very rarely used and you can just do some construction:
[image]

Maybe we should allow adding an angle constraint between three points, but even that seems extraneous.

whitequark, 2016-04-16:
Also, let's say we do want to specify an angle over 180° degree. Then the dimension line should look something like this (pardon my crude sketching):
[image]

So there's also no ambiguity.

Evil-Spirit, 2016-04-16:
> Do we need that? It's very rarely used and you can just do some construction:

So, try to constrain 100 arcs by involving additional segments. There will be churn.

[image]

Evil-Spirit, 2016-04-16:
https://youtu.be/b41zpAzhH_M?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT

whitequark, 2016-04-16:
The video contains a very artificial example. People don't actually use parametric modeling like that, it's a job more suited to Blender or something. And overall, angles over 180° are extremely rare; in the sketch you posted above, the dimension used to constrain the inner flat would be the distance to center, or perhaps distance to opposite side of the arc.

Evil-Spirit, 2016-04-16:
Anyway, we have to do the best way if we know how. The degrees > 180 are useful, angle for arcs are useful (without any additional sketching), and arc lengths are useful (there will be the same visualization as angle).

whitequark, 2016-04-16:
I don't think they are useful, I don't see any compelling use cases, and as such implementing them is a waste of time. And regardless of that there isn't any ambiguity anyway as I've shown above.

whitequark, 2016-04-16:
> there isn't any ambiguity

Specifically, the angle being constrained/measured is indicated by a reference arc across that angle. You don't get any less ambiguous. The label may occlude it entirely but it's still unambiguous because there is no reference arc on the *other* side.

Evil-Spirit, 2016-04-16:
There is no reason why we should'nt go thorugh the footpath I already pass. It's time already wasted, so we can have it in much less time. The sooner or later. The small things like this doing all the imression. Devil in the details. etc.

whitequark, 2016-04-16:
We have to add that to the file format, maintain it, fix bugs in it, etc. I would rather not do all that for a useless feature. If a feature has no compelling use case, I don't care how easy it is to add it, I won't add it, because features used by no one stay alive for years anyway and extract their maintenance cost.

Evil-Spirit, 2016-04-16:
@whitequark , 
@jwesthues ,
there is no specific; just gui work. anyway we MUST do atan2 angle in order to support 	chirality(enantiomorphism) of the sketches. Just look at ugly workaround to hide acos(dot) angle equation problems:
```
                // avoid false detection of inconsistent systems by gaining
                // up as the difference in dot products gets small at small
                // angles; doubles still have plenty of precision, only
                // problem is that rank test
                Expr *mult = Expr::From(arc > 0.99 ? 0.01/(1.00001 - arc) : 1);
```

whitequark, 2016-04-16:
Yes, atan2 will need to be done, but that happens after refactoring and once we have migrated to a new file format, as discussed before.

Until then, would you please focus on the particular issue and fix the arrows? It's the last remaining part right now...

Evil-Spirit, 2016-04-16:
[image]

whitequark, 2016-04-16:
Excellent, thanks! Fixed in a4a353f01.

closed 2016-04-16 by whitequark