whitequark/solvespace #96: Stipple line type ignored when try export to PDF or other

Symbian9, 2016-03-24:
When I try export drawing that include dashed lines or other types its always export only as it was single type.

whitequark, 2016-04-08:
PDF/EPS done in 94cba11.

Symbian9, 2016-04-08:
Stippled lines in 2016.04.08 nightly build have an issues:
Testing - https://github.com/whitequark/solvespace-testcases/tree/master/tests/stippled

1) if export scale 1:1.0000
* 1st line hase 10 segments instead to 5 (but in SolveSpace I see really 5, not 10). All other styles also hase doubled number of needing segments

2) if export scale 1:0.500
* all Ok, first line and all other have same number as it described in comments

3) if export scale 1:2.000
* all Ok, first line and all other have same number as it described in comments

Evil-Spirit, 2016-04-08:
Please, report additional info about file type and viewer

Evil-Spirit, 2016-04-10:
@Symbian9, please, say which filetype, os, and viewer do you use.

whitequark, 2016-04-10:
SVG done in 4a748166e.

closed 2016-04-10 by whitequark