whitequark/solvespace #97: Export to DXF propose name "untitled.AutoCAD 2007"

Symbian9, 2016-03-24:
When I try export DXF File  propose me for resulted file name "untitled.AutoCAD 2007" and not "untitled.dxf" as it must be.

Evil-Spirit, 2016-03-25:
Which operating system do you use?

whitequark, 2016-03-25:
This works for me on Linux.

Evil-Spirit, 2016-03-25:
I have no problems on Windows too.

Evil-Spirit, 2016-03-25:
@Symbian9 please, specify steps to reproduce this bug.

Symbian9, 2016-03-25:
@Evil-Spirit this is simple
1) run SolveSpace and draw or not draw nothing
2) File -> Export 2D View
3) Choose format "DXF File (AutoCAD 2007) (*.dxf)"
RESULT: it propose "untitled.AutoCAD 2007" name.

As I see in previouse builds format named as "DXF File (*.dxf)" and it work normally.
Think main problem is in double brackets that now presented, and it use content only from first brackets.
Maybe need rename it to "DXF File, AutoCAD 2007 (*.dxf)" or "DXF 2007 File (*.dxf)", where only one brackets with format extension end.

whitequark, 2016-03-25:
Well, this is not reproducible on either Windows or my Debian. So without more details about your system I can't fix it.

Symbian9, 2016-03-25:
I have Linux Mint 13, 64bit. 

But anyway think its problem not in OS but in SolveSpace code where extension ends added to file name.
Previouse builds of SolveSpace where format named as <code>"DXF File (*.dxf)"</code> add only "dxf" extension to file name.
Is it really needed use <code>(AutoCAD 2007)</code> in rounded brackets in format name?
As I see, there no similar double rounded brackets in LibreCAD, where formats in saving dialog named as <code>"Drawing Exchange DXF 2007 (*dxf)"</code>
In Inkscape sew that they use squared brackets in format description, but also not double rounded brackets
<code>"HP Graphics Language Plot file [AutoCAD](*.plt)"</code>

whitequark, 2016-03-25:
Fixed in 6f67ec2.

closed 2016-03-25 by whitequark

Symbian9, 2016-03-25:
Also if you can rename it to "DXF 2007 File (*.dxf)" in format choosing dialog.

whitequark, 2016-03-25:
I don't see any point in that.

Symbian9, 2016-03-25:
So, its steal "DXF File (AutoCAD 2007) (*.dxf)"?
But its really look buggy.

whitequark, 2016-03-25:
It's not.

Symbian9, 2016-03-25:
Why then not "AutoCAD DXF 2007 file (*.dxf)"?
I often test different software and dont yet sew this "name(one)(two)" naming scheme in format descriptions in other software export or save dialogs.

whitequark, 2016-03-25:
I don't care what other software does, and I'm not doing change for sake of change.