whitequark/solvespace #98: Improve Three.js controls

whitequark, 2016-03-25:
Need to integrate the changes by @cr1901 here: https://github.com/cr1901/solvespace.js

cr1901, 2016-03-25:
I will write a script to automate export (to the extent possible). Thoughts on linking my repo as a submodule so that we have an easy ID to figure out which version of the code is in Solvespace? Or perhaps as a post-commit hook, I'll stamp the code with the version in case bug fixes are needed.

whitequark, 2016-03-25:
Nah, no submodules, too complex. I'll write an asset system for SolveSpace at some point, but not yet.

cr1901, 2016-03-25:
Okay cool, I'll just make a post-commit hook to stamp the classes with a version ID variable for now. So if you or anyone has any bugs to report, I can figure (more) quickly which commit to look at.

whitequark, 2016-04-16:
@cr1901 This is completely broken. Dragging the left mouse button does not rotate the model, and right mouse button now brings up a popup menu instead of dragging the model.

cr1901, 2016-04-16:
@whitequark As a consequence of removing OrthographicTrackballControls, middle mouse click now rotates the model to match Solvespace behavior. Want me to change it back to the way it was?

This idea was if you right click without moving the mouse at all a context menu will be brought up in case you still needed it. I must've implemented this improperly, so I'll get rid of it.

Evil-Spirit, 2016-04-16:
@cr1901 I think we have to consider touch-screen cases, so the SolveSpace's behaviour seems to be useless.

cr1901, 2016-04-16:
@Evil-Spirit Whitequark and I have been talking on IRC, and I agree. This was an accessibility fail for me.

Even without considering touch events, the controls as-is only work properly with a USB mouse (which is what I primarily use- oops!).

whitequark, 2016-04-16:
All seems OK, testing now.

whitequark, 2016-05-06:
Fixed in affc88f3.

closed 2016-05-06 by whitequark