whitequark/solvespace #99: 2016-03-18 nightly doesn't open on el capitan

aljex, 2016-03-28:
Just tried to run the current nightly for osx, and it fails to open because it can't find:
/opt/X11/lib/libfreetype.6.dylib

---quote---
Process:               solvespace [3267]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [3267]
User ID:               501

Date/Time:             2016-03-27 22:25:56.969 -0400
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        08BBAC0F-488E-D237-CE08-9628474EC259

Time Awake Since Boot: 6600 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /opt/X11/lib/libfreetype.6.dylib
  Referenced from: /Applications/solvespace.app/Contents/MacOS/solvespace
  Reason: image not found

Binary Images:
    0x7fff5ff83000 -     0x7fff5ffba007  dyld (360.19) <9D05FDF4-65CE-3B53-86D4-ABE1A5BF35F3> /usr/lib/dyld
    0x7fff83edd000 -     0x7fff83eebfff  com.apple.opengl (12.0.40 - 12.0.40) <D37D88AE-0731-3105-99AB-984DA9B9BCE6> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL

Model: MacBookAir7,1, BootROM MBA71.0166.B09, 2 processors, Intel Core i5, 1.6 GHz, 4 GB, SMC 2.26f2
Graphics: Intel HD Graphics 6000, Intel HD Graphics 6000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 27.2
---end---

altaic, 2016-04-02:
Homebrew installs to `/usr/local/Cellar/<package>/<version>/lib` and symlinks to `/usr/local/lib`. My guess is that you don't have libfreetype in the path it expects. Try `ls -l /opt/X11/lib/libfreetype.*`.

Two workarounds:
- Symlink it `ln -s /usr/local/lib/libfreetype.6.dylib /opt/X11/lib/libfreetype.6.dylib`.
- Install [XQuartz](http://www.xquartz.org)

whitequark, 2016-04-02:
Hm, I was under impression that freetype is a standard OS X component. I guess I can bundle it in the `.app` if it isn't. Is it not?

altaic, 2016-04-03:
I don't believe so. In any case I'm pretty sure `/opt` doesn't exist by default, so linking to anything in there will run into trouble. The best policy is to bundle all libraries with the app, unless the libraries are part of the OS X SDK, i.e. they're in `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/lib/`.

whitequark, 2016-04-03:
Ah! Okay. I've never really used OS X. I will fix this.

whitequark, 2016-04-08:
I believe this is fixed in the latest nightly. Please reopen if it is not.

closed 2016-04-08 by whitequark

aljex, 2016-04-08:
I'll try it when I get a chance. Thanks!

aljex, 2016-04-10:
App opens with no error now. Thank you!