whitequark/solvespace #100: Point-on-line solver issue

whitequark, 2016-04-02:
To reproduce:
  1. Draw a circle.
  2. Extrude.
  3. Add a datum point in 3d.
  4. Constrain datum point to the segment created from the circle center.

Solver fails. It shouldn't.

Evil-Spirit, 2016-04-07:
Don't want to go deeper without solver debugger. I suggest to implement LaTeX debug output (since MathML is dying).

whitequark, 2016-04-08:
OK. Let's do LaTeX debug output.

Evil-Spirit, 2016-04-08:
Or we can use MathJax.

whitequark, 2016-04-08:
I have no opinion on what tool to use here.

whitequark, 2016-04-08:
Does this fail too with Jonathan's solver, or did we break it?

Evil-Spirit, 2016-04-08:
Jonathan's version also has the same issue.

jwesthues, 2016-04-09:
That works for me if the length of the line segment is constrained, or if the point and point-on-line are in a later group. Otherwise, it fails for me by choosing a stupid (zero) extrusion depth, at which point the line degenerates to a point and the point-on-line constraint fails.

So I think the problem here is just that "extrusion depth" lacks the same kind of heuristics to choose reasonable solutions when underconstrained as we have when solving just for entity geometry, not anything specifically with that line or constraint. You can probably find similar misbehavior with step and repeat or assembly groups, for similar reasons.

Evil-Spirit, 2016-04-10:
@jwesthues, we can rewrite PT_ON_LINE equation like:
`p = lp0 + (lp1 - lp0) * t`
where:
p - point
lp0, lp1 - line points
t - value of constraint(valA)
so, we can eliminate AT_MIDPOINT (this will be PT_ON_LINE with valA = 0.5)

jwesthues, 2016-04-10:
PT_ON_LINE constrains 1/2 DOF in 2d/3d, but AT_MIDPOINT constrains 2/3. For the equation that you write to constrain "point on line", the parameter t must be arbitrary, not fixed to valA.

For the PT_ON_LINE case (but not the AT_MIDPOINT case), you could introduce a new solver parameter for t, but the result seems similar to before but slower, more parameters in the O(n^3) solve. You could project that equation onto the 1 (2d) or 2 (3d) basis vectors perpendicular to the line, but that's basically what we're doing now.

I also don't see how that would stop the solver from driving the unconstrained extrusion depth to zero and blowing up, which is the problem here. What are you trying to do?

Evil-Spirit, 2016-04-10:
Actually i dont know how it fix the problem, but its need to try,

jwesthues, 2016-04-10:
On investigation, I (a) confirm that the problem is the unconstrained extrusion depth going to zero, and (b) retract my earlier statement blaming extrude and other similar non-entity parameter treatment for this problem; you can easily reproduce it with a point in 3d constrained to lie on any line whose length is free.

PT_ON_LINE is written using VectorsParallel, for no obvious reason. Rewriting that constraint to work on two projected distances (using any two basis vectors perpendicular to the line) should fix that problem, since replacing the "point on line in 3d" constraint with two "point on line in 2d" constraints works. That still has the hairy ball problem of choosing the basis vectors, which you can't do with a continuous function; you'd need Vector::Normal() or equivalent.

You could write three equations and make the constraint itself introduce one new parameter for t. I don't know how well that would work numerically, but it would avoid the hairy ball problem, perhaps elegant at the cost of speed.

Evil-Spirit, 2016-04-11:
@jwesthues, also, we can use the 3d point-line distance expression:
`length(cross(p0 - p1, p0 - p)) / length(p0 - p1) = 0`

jwesthues, 2016-04-11:
That constrains one DOF, and point-line-distance is in fact coded that way already. You still need to constrain one more DOF for point-on-line, so you need one more equation.

Evil-Spirit, 2016-10-22:
@jwesthues, 
I have some experiments about 3d-parallel constraint and figured out what we can also write it in form, which I call "direct-expressed dof transformation". The main idea is to fully constrain dofs we have before and express remaining dofs in new parameters. 

For parallell constraint of two segments we take 3 dof and introduce 1:
l0p0, l0p1 - segment 0 (vector)
l1p0, l1p1 - segment 1 (vector)
r - ratio - introduced degree of freedom (scalar free parameter)
```
d0 = l0p0 - l0p1
d1 = l1p0 - l1p1
d0 = d1 * r
```
So we have resulting dof decreased by 2, as we want.
Also, we get albility to constrain r > 0 or r < 0 to have parallel or antiparallel constraints.
@jwesthues, don't bother about complexity of the solver, there is linear-eq solvers with decreased complexity like operating with sparce-matrices. I have tested and this can be applicable (realtime) until 5000x5000 matrices (5000 eq + 5000 params). This is enough, I think.

jwesthues, 2016-10-22:
By introducing an extra parameter, we can indeed avoid the hairy ball problem, as noted above. There's perhaps some analogy between that and the use of unit quaternions (by four parameters the magnitude constraint) for a rotation's three DOF.

5000x5000 seems surprisingly good. What kind of constraints were you testing? And do you mean solution of the linear system by iterative methods, or by stuff more analogous to normal LU factorization, just with sparse data structures? Though I'd agree that a sparse solver is generally worth investigating, since that's potentially a significant "drop-in" speedup.

Evil-Spirit, 2016-10-22:
@jwesthues,
Some of stress-test of constraints you can see here
https://youtu.be/soMZPMSpdkQ?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT
https://youtu.be/cjxGTqps884?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT
https://youtu.be/eSGlx2UW84w?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT
https://youtu.be/O_wa9xU8xs4?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT
https://youtu.be/soMZPMSpdkQ?list=PLqmaFsEpxcUXzvoLdZ7QtpKb_iInqo1pT
The library used for linear system solving is Eigen. I have tested some methods from Eigen. QR decomposition on sparse matrices seems to be best of all, but I have not enough knowledge about all the methods and can setup wrong parameters and because of this have no such success with them.

Evil-Spirit, 2016-10-22:
@jwesthues,
Iterative methods seems very bad for this application. Using them causes increasing of Newton-method steps number. There are better results than QR in speed, but they are more unstable or have issues in Eigen implementation.

Evil-Spirit, 2016-10-22:
@jwesthues, all this demos recorded on the laptop with i7 processor (http://www.notebookcheck.net/Intel-Core-i7-3610QM-Notebook-Processor.72681.0.html)

Evil-Spirit, 2016-10-22:
For done this we need:
1. Make ability to generate parameters by constraints.
2. Make ability to guess valA parameter (something like ModifyToSatisfy)
3. Ensure all the legacy data loading the right way

Evil-Spirit, 2016-11-01:
@whitequark 
@jwesthues 
Here is prototype for rewritten equations for
1. pt-on-line https://github.com/Evil-Spirit/solvespace/commit/25a193ff4ec1548231e47715bbf5453191c28eeb
2. parallel https://github.com/Evil-Spirit/solvespace/commit/d511f7a15009264fa5fa17a91f2719fd733a9e87

Evil-Spirit, 2016-11-06:
@jwesthues,
@whitequark,
I have checked jasen-asm with new equations and it seems problems with redundant has gone.
New equations not very robust on dragging, but an old ones based on VectorsParallel is more ugly. I propose to get rid of VectorsParallel logic everywhere.

whitequark, 2016-11-06:
> New equations not very robust on dragging

That seems like a regression, and I don't think we should replace the old logic if it works but is ugly, with new logic that works worse.

Evil-Spirit, 2016-11-06:
@whitequark, nothing says about regression, just strange behaviour. I could say it in other words: the old equations is not ugly, it just completely wrong. I think we have to change them anyway.

Evil-Spirit, 2016-11-06:
btw, pt-on-line working pretty good. Parallel causes the troubles: we have asymetrical reaction on dragging of two identical segments.

whitequark, 2016-11-17:
> @whitequark, nothing says about regression, just strange behaviour. I could say it in other words: the old equations is not ugly, it just completely wrong. I think we have to change them anyway.

Okay, can you please show some examples of:
  * completely wrong behavior of old equations, and
  * where the new equations behave oddly?

I'd really like to understand this change.

Evil-Spirit, 2016-11-17:
@whitequark,
You can found strange behaviou only when you drag: you cannot reverse one of parallel lines, but you can reverse the second one. The same situation with the pt-on-line. Anyway, we must cange it only for 3d cases, because the current equations can be easily written for 2d with no any problems (atan2 for parallel and pt-line-distance == 0 for pt-on-line).

whitequark, 2016-11-17:
> you cannot reverse one of parallel lines, but you can reverse the second one. The same situation with the pt-on-line

Ah, this does not matter at all because we want chirality-preserving constraints anyway! I'll test this some more and merge.

Evil-Spirit, 2016-11-17:
@whitequark, yes, you are right, this matters only when we want an old-sytle behaviour.

Evil-Spirit, 2016-11-17:
@whitequark 
Also, you should consider what I haven't implemented legacy files support, so if you will decide to merge, tell me and I will insert some legacy-support code.

whitequark, 2016-11-17:
@Evil-Spirit Why do we need any changes at all to support old files?

Evil-Spirit, 2016-11-17:
@whitequark,
Because we introduce new feature: parameters, generated by constraints. So we need to create them and set initial values for old files.

whitequark, 2016-11-17:
@Evil-Spirit Is this like the problem with additional points for text entity?

Evil-Spirit, 2016-11-17:
@whitequark, yes, we should go the same way as for text.

whitequark, 2016-11-17:
OK

Evil-Spirit, 2016-11-19:
@whitequark 
So, have you tested this? Should I implement legacy support?

whitequark, 2016-11-20:
@Evil-Spirit yes

Evil-Spirit, 2016-11-21:
@whitequark,
pt-on-line here https://github.com/Evil-Spirit/solvespace/commit/9c98f5ca785fa5e3a3bd5dfd41b6a3c6cbb8acb3

whitequark, 2016-11-26:
@Evil-Spirit Merged, with some modification to make the code fit better. Now the vector parallel case.

Can you show some example where existing vector parallel code is broken? And rebase that commit on top of master?

Evil-Spirit, 2016-11-26:
@whitequark, okay

Evil-Spirit, 2016-11-26:
@whitequark 
rebased
https://github.com/Evil-Spirit/solvespace/commit/ccb002b6e5d1746ab6a0f9a94a746b24074808bf

Evil-Spirit, 2016-11-26:
@whitequark,
I have tested jasen-asm and it was fixed by pt-on-line, so I have no testcases for vector parallel.

whitequark, 2016-11-26:
@Evil-Spirit Parallel constraint merged except two parts:
  1. Move of UpgradeLegacyData to OpenFile (nothing obviously wrong with it, I just want to understand why)
  2. Revert of 3a585ea7, let's do that when VectorsParallel is no longer used at all.

Evil-Spirit, 2016-11-27:
1. We have to do it (see code comments)
2. I think we should only remove the lines, we need SolveGroup without reporting for other improvement (https://github.com/whitequark/solvespace/issues/208)

whitequark, 2016-11-27:
> I think we should only remove the lines, we need SolveGroup without reporting for other improvement (#208)

But what about VectorsParallel()? Doesn't same orientation constraint have the same issue, potentially?

Evil-Spirit, 2016-11-27:
@whitequark I am almost get rid of this

Evil-Spirit, 2016-11-27:
@whitequark https://github.com/Evil-Spirit/solvespace/commits/master-parallel-2
I am working on the last SAME_ORIENTATION

Evil-Spirit, 2016-11-27:
@whitequark 
done here https://github.com/Evil-Spirit/solvespace/commits/master-parallel-2

whitequark, 2016-11-27:
@Evil-Spirit I merged everything except "Move of UpgradeLegacyData to OpenFile", because that fails lots of tests.

Evil-Spirit, 2016-11-27:
@whitequark, ok I will check it

Evil-Spirit, 2016-11-28:
@whitequark 
fixed properly
https://github.com/Evil-Spirit/solvespace/commits/fix-upgrade-legacy

whitequark, 2016-11-28:
Merged

closed 2016-11-28 by whitequark