whitequark/solvespace #101: Notes of SolveSpace 2.1 on Libre Graphics Meeting 2016

Symbian9, 2016-04-02:
Please, make progress notes for upcoming biggest meeting of opensource multimedia apps developers.
http://libregraphicsmeeting.org/2016/
<i>"You can still send in slides for The State of Libre Graphics Meeting. On site you can sign up for 5m lightning talks."</i>
http://libregraphicsmeeting.org/2016/state-of-libre-graphics-union-2016/

Evil-Spirit, 2016-04-04:
What's the point in taking part?

Evil-Spirit, 2016-04-04:
@Symbian9, What's the point in taking part?

Symbian9, 2016-04-04:
@Evil-Spirit, think, present progress on current development and present project itself on biggest conference about opensource graphics software is itself good target, because at this moment SolveSpace steal not very wide known.
Before 8 april its possible add only small progress notes that will be included in one general presentation.

whitequark, 2016-04-08:
Ah, I haven't noticed this early enough. Too late now anyway.

whitequark, 2016-04-08:
But I'll consider submitting a talk to some Libre Graphics conference in the future...

closed 2016-04-08 by whitequark