whitequark/solvespace #106: DXF import

whitequark, 2016-04-10:
Much-requested feature: http://solvespace.com/forum.pl?action=viewthread&parent=24 and not too hard with the new DXF library. Let's do it in 2.1.

Evil-Spirit, 2016-04-10:
Ok, its no problem with it.

whitequark, 2016-04-10:
In the menu we can put it under `File → Import...`.

Symbian9, 2016-04-10:
@whitequark, you mean "Menu -> New Group -> Import / Assemble..."?

whitequark, 2016-04-10:
No. That's for linking other files and tracking changes in them. This DXF import feature will not do that.

Evil-Spirit, 2016-04-12:
@Symbian9, I don't think so, the svg is pure graphical format, there is no any usecase when it can be applicable.

whitequark, 2016-04-12:
No, I'm not interested in SVG.

Evil-Spirit, 2016-04-14:
[image]

Evil-Spirit, 2016-04-14:
[image]

Symbian9, 2016-04-14:
@EvilSpirit, COOL! When i can test this nightly?

Evil-Spirit, 2016-04-14:
When i done.  You can send me dikes which you want to import

Evil-Spirit, 2016-04-14:
When i done. You can send me files which you want to import

Symbian9, 2016-04-14:
@Evil-Spirit, did you try make sketch -> export to DXF -> open in LibreCAD & resave in other DXF -> and then import it to SolveSpace again?

Evil-Spirit, 2016-04-14:
Round-trip
[image]

Symbian9, 2016-04-15:
@Evil-Spirit, what going with circles and arcs after this "round trip"?
As I can see, there some segmented circles or you did some settings for chord tolerance?

Evil-Spirit, 2016-04-15:
@Symbian9, this is just because chord tolerance, one of acrs has very far-distanted center-point, so the chord tol is too big.

Symbian9, 2016-04-15:
@Evil-Spirit, no. Circle in SolveSpace include only 1 point at center.
On your "round-trip" screen I see circle with 5 points, so something wrong with it!

Evil-Spirit, 2016-04-15:
Circles just becomes an arcs. This is an export issue, not the import.

Evil-Spirit, 2016-04-15:
@Symbian9, just learn how to build it yourself.

Evil-Spirit, 2016-04-15:
Sample with splines:
[image]

whitequark, 2016-04-16:
@Evil-Spirit Are you done with this, or is there something else?

whitequark, 2016-04-16:
@jwesthues Where do you think the menu item should be? I feel like we need to distinguish the existing `Import / Assemble` command which does a linked import, and this new command, which does a non-linked import. I think it should be put under `File` as `Import...`, and `Import / Assemble` should be renamed to `Link / Assemble` to better describe the function that it performs. I think it also makes sense, in the future, to do allow to link in DXFs.

jwesthues, 2016-04-16:
All of that make sense to me. Linked DXF import is potentially a rather complicated problem, since for any constraints against the linked entities to work usefully, you'd need heuristics to consistently assign identifiers to the imported entities.

whitequark, 2016-04-16:
> you'd need heuristics to consistently assign identifiers to the imported entities.

Yeah. The idea was to use the handles in DXF files to generate entity IDs. Handle reuse is a problem, but then it is also a problem for linking sketches...

Evil-Spirit, 2016-04-16:
@whitequark, this depend of how deep we have to go.
There is basic entities imported;

Need to import:
* text
* textstyles

What to do with:
* dimensions
* blocks
* layers
* ltypes

whitequark, 2016-04-16:
> blocks

For now: expand them. Expand them inline every time they're used.

> layers, ltypes

How about this: take the entire set of styles we care about from both the layer and the entity, merge them, and squash it into a string. E.g. `0.1-continuous-ff0000` or something like that. Then either create a new custom style with that name or reuse one if it already exists, even if the style has different actual attributes--this way it would be easy to remap attributes on several DXF files imported one after another.

> dimensions

When we can determine exactly that a diameter or an angular dimension constrains specific points, then a constraint should be created for those. When we cannot, increment a counter and then display a message saying "n constraints from the imported file were discarded".

But on top of that, we should infer the following constraints from *geometry*:

  * Point-coincident
  * Horizontal
  * Vertical

All of the above are unambiguous and they won't result in sketches that can't solve, so we can safely do that. And it will be very convenient if one wants to maintain the drawing.

Evil-Spirit, 2016-04-16:
@whitequark,
I thiking about autoconstraining coincident, horizontal and vertical (optional) and about dimensions which is exactly fit the entities, so we can constrain.

Evil-Spirit, 2016-04-16:
@jwesthues , We really have the problem with solver which throws params not envolved in equations. 
Why we should do it? We just have no any constraints, but the solver overdriven (max params exceeded).

jwesthues, 2016-04-16:
No reason beyond laziness; I'd intended SolveSpace for typical parametric modeling, where all (or almost all) parameters are constrained and this doesn't matter much. It makes sense to exclude parameters not mentioned in any equations from the Jacobian, a slight speedup even when you don't exceed `MAX_UNKNOWNS`.

Symbian9, 2016-04-17:
Some DXF samples for testing
http://cadkit.blogspot.nl/p/sample-dxf-files.html

Evil-Spirit, 2016-04-21:
@whitequark, how to commit libdxfrw?
```
    DRW_Spline(const DRW_Spline& p):DRW_Entity(p){
        eType        = DRW::SPLINE;
        normalVec    = p.normalVec ;
        tgStart      = p.tgStart   ;
        tgEnd        = p.tgEnd     ;
        flags        = p.flags     ;
        degree       = p.degree    ;
        nknots       = p.nknots    ;
        ncontrol     = p.ncontrol  ;
        nfit         = p.nfit      ;
        tolknot      = p.tolknot   ;
        tolcontrol   = p.tolcontrol;
        tolfit       = p.tolfit    ;
        
        for(double v : p.knotslist) knotslist.push_back(v);
        for(double v : p.weightlist) weightlist.push_back(v);
        for(DRW_Coord *v : p.controllist) controllist.push_back(new DRW_Coord(*v));
        for(DRW_Coord *v : p.fitlist) fitlist.push_back(new DRW_Coord(*v));
    }
```

whitequark, 2016-04-21:
Just fork it, like with solvespace.

Evil-Spirit, 2016-04-21:
@whitequark, may be just direct push into branches?

whitequark, 2016-04-21:
Not for now.

whitequark, 2016-04-21:
Specifically, I see code review as necessary before landing changes to `master`, and if you don't push into `master` then what difference does it make which repo it is?

Evil-Spirit, 2016-04-21:
> push into branches
not master

i just don't know what is happening when i switch the main repo after switching submodule

whitequark, 2016-04-21:
Ah... The submodule is, logically, a file with a commit hash. Once you commit it, it will stay at that hash, but other people won't be able to check out the submodule unless they do `git fetch git://github.com/evilspirit/libdxfrw` first.

whitequark, 2016-04-21:
Let's work with forks for now. Once we get 2.1 out I'll think about switching to a more convenient workflow, now that we have the solvespace organization.

Evil-Spirit, 2016-04-22:
What about making points like this?
[image]

whitequark, 2016-04-22:
let's discuss it on IRC.

Evil-Spirit, 2016-04-24:
> Some DXF samples for testing
http://cadkit.blogspot.nl/p/sample-dxf-files.html

@Symbian9, Thank you for useful examples. Can you say what kind of license it use and can i commit this into testcases repo? BTW, is there some dxf test-tables with all types of entities and all the parametes and cases used?

Symbian9, 2016-04-24:
I found this link in at the end of README.md from LibreCAD
https://github.com/LibreCAD/LibreCAD_3/blob/master/README.md

Try ask it on LibreCAD project page, because I don't know what is the license

whitequark, 2016-04-26:
@Evil-Spirit What's left here? Just automatically adding constraints?

Evil-Spirit, 2016-04-26:
@whitequark, I almost done with constraints.

Evil-Spirit, 2016-04-26:
[image]

Evil-Spirit, 2016-04-26:
with autoconstraining(H/V, PC)
[image]

Evil-Spirit, 2016-04-26:
[image]

whitequark, 2016-04-26:
Looks wonderful!

Symbian9, 2016-04-27:
@Evil-Spirit, why all around in red?

Evil-Spirit, 2016-04-27:
@Symbian9, solver can't create matrix for such big (MAX_UNKNOWNS exceeded). 
> No reason beyond laziness; I'd intended SolveSpace for typical parametric modeling, where all (or almost all) parameters are constrained and this doesn't matter much. It makes sense to exclude [...]

Evil-Spirit, 2016-04-27:
@jwesthues,
But if we could remove problem with solver overdrive by params, not envolved in equations, we can make linear algorithm for SolveBySubstitution, and because H/V PC constraints not envolved into solving, we can solve sketches any size even with MAX_UNKNOWNS.

whitequark, 2016-05-07:
Done in 70d84b3 and a21a327a, pending some details in #149.

closed 2016-05-07 by whitequark