whitequark/solvespace #109: Assign to other style measure constraints

Symbian9, 2016-04-10:
For now some special marks (as "H", "V" and few other) there really no need to export in 2D Drawing.
But its impossible hide them separatly from other like angle constraint and length that preferable for printing. 
Think, it would be good, if it would be possible assign to other style some of constraints.
Also, in future it will help setup measure look exactly as drawing standarts
[image]
http://www.me.umn.edu/courses/me2011/handouts/drawing/blanco-tutorial.html

whitequark, 2016-04-10:
In the future SolveSpace will have a dedicated drawing module, which will solve this problem and others. However, we are many months away from this, so closing for now.

closed 2016-04-10 by whitequark

Evil-Spirit, 2016-04-10:
We could support something like dxf dimstyle in order to support any standart (iso, ansi, gost, etc)

Symbian9, 2016-04-10:
@Evil-Spirit, it would be cool if SolveSpace will have inbuild ISO and GOST technical and mechanical drawing standarts.
But anyway, some standarts can change from time to time, so need some advanced futures.

Evil-Spirit, 2016-04-11:
Actually we can solve it by allowing styling the constraints