whitequark/solvespace #110: Stop using 80-symbols guideline.

Evil-Spirit, 2016-04-13:
@whitequark,
@jwesthues,
I have the half of my sceen empty;
Why we need to do ugly line-breaks?
Could you enlarge your consoles under the linux?
[image]
[image]
[image]
[image]

jwesthues, 2016-04-13:
That, I care less about. I very slightly prefer 80-column lines, but at your/whitequark's discretion (though I wouldn't count your second example as ugly, since it aligns two analogous things).

Evil-Spirit, 2016-04-13:
Sorry, could we use at least 80 symbols before newline-breaking?

jwesthues, 2016-04-13:
I mean if whitequark doesn't mind wider, I'm good with it.

Evil-Spirit, 2016-04-13:
There is not so much cases when it really helps (like aligning inside function calls), but this is so ugly inside functions-prototypes.

jwesthues, 2016-04-13:
If you want to go wider for new code (or as part of other changes) and whitequark doesn't mind, I don't object. Please don't make changes for formatting of existing code only, since that creates low-value churn.

Evil-Spirit, 2016-04-13:
Ok, of course just for new or edited code;

whitequark, 2016-04-13:
Is 100-characters ok? This is what I generally use for my other code.

whitequark, 2016-04-13:
FWIW I don't believe in console editors. I'm using Sublime Text on a widescreen laptop and fitting 120 columns makes text a bit too small to comfortably read.

Evil-Spirit, 2016-04-13:
ok, lets do 100.

closed 2016-04-13 by whitequark