whitequark/solvespace #112: Assemble polylines when exporting into DXF

whitequark, 2016-04-15:
There's a draft at 8cbf2d6715de, but it's quadratic. Investigate.

Evil-Spirit, 2016-04-15:
@jwesthues, @whitequark , Should we introduce SkdTree for BBox or change existent SkdTree to operate with BBoxes?

jwesthues, 2016-04-15:
So why a k-d tree of bounding boxes? You don't need to consider the point comparison tolerance when you build the tree, as long as you consider the tolerance when you traverse the tree later. It seems like all you really need is a k-d tree of points (not yet implemented), or a k-d tree of edges (implemented as `SKdNodeEdges`) at the cost of some efficiency.

Evil-Spirit, 2016-04-15:
When i say about KD-tree based on bound boxes, i mean the universal variant of k-d tree for points, edges and triangles just for not to have N implementations. In the future we would need Entity k-d to improve visualization and hitting perfomance, etc.

Evil-Spirit, 2016-04-15:
@jwesthues, so what you think about?

jwesthues, 2016-04-15:
Common seems reasonable to me. It's already kind of stupid that e.g. the heuristics for deciding how to split and when to give up and make a leaf node are duplicated, and they'd end up triplicated if you did a separate k-d tree of points for this.

A base k-d node class with two points for the bounding box, and then point, edge, and triangle node classes that inherit? That would replace both `SKdNode` and `SKdNodeEdges`.

Evil-Spirit, 2016-04-15:
I think it can be template with number of dimensions and a type of containter. We can just implement GetBBox function for each thing we want to store. Or we can use external functions GetBBoxOf([Vector | Edge | Triangle | Entity | Constraint | etc]) to use inside template code.

whitequark, 2016-04-15:
```
template<class T> class SKdNode { ... T t; ... t.GetBBox(); 
```
?

jwesthues, 2016-04-15:
Templated seems good too.

whitequark, 2016-04-15:
Also, do we need to parameterize it over number of dimensions? Isn't everything we have here in 3d? Seems like overengineering to me.

Evil-Spirit, 2016-04-15:
We need pure 2d (for export). But this is just kind of optimization which we can do for now (for less time), and hard(for greater time) to do in the future when we really need it .

Evil-Spirit, 2016-04-15:
`template<class T> class SKdNode { ... void Add(T t) { t.GetBBox();`
or
`template<class T> class SKdNode { ... void Add(T t) { GetBBox(t);`

whitequark, 2016-04-15:
Can we do something simple to parameterize SKdNode over number of dimensions e.g. parameterize it over a vector type? Otherwise this seems like a lot of added complexity ot me.

Former, `t.GetBBox()`. Fewer global functions etc.

jwesthues, 2016-04-15:
The speed penalty for 2d items in a 3d tree should be small when it's built (since we'd consider but always reject splits by z), and ~zero when it's traversed; so I probably wouldn't bother. I don't see a particularly clean way to parametrize the vector type.

Evil-Spirit, 2016-04-15:
We have to use Vector and will get size overhead. But we will get calculation boost just to not considering z values. In future we can do Vector template-parametrized to use it propely.
t.GetBBox() is good,
but GetBBox(t) is flexible (we can do specific functions for external-library things without modifing it's code). 
Anyway, there is no difference at now.

Evil-Spirit, 2016-04-15:
@jwesthues, ok, wouldn't bother about it. Just 3d case.

Evil-Spirit, 2016-04-20:
@jwesthues, 
What the point of this calculations? Can you explain?
`badness[i] = pow((double)ltc[i], 4) + pow((double)gtc[i], 4);`
may be you mean
`badness[i] = fabs(pow((double)ltc[i], 4) - pow((double)gtc[i], 4));`?

jwesthues, 2016-04-20:
A split can be "bad" in two different ways: it can be uneven (many more items in one branch than the other), or it can be useless (many items in both branches; remember that an item that intersects the split plane gets pushed down both). That metric captures both effects. The x^4 is pretty arbitrary, but higher powers penalize unevenness more.

For 8 items, your modified metric would treat a 4:4 (i.e., perfect) split identically to an 8:8 (i.e., perfectly useless; the plane intersects all items) split. That seems very bad.

whitequark, 2016-04-21:
Aleksey has reimplemented DXF polyline reassembly using a spatial hash table, which in my opinion is quite elegant and certainly very simple.

Fixed in 59ebf22f.

closed 2016-04-21 by whitequark