whitequark/solvespace #113: Don't display `x.00` for integer distances and angles

whitequark, 2016-04-15:
@jwesthues What's your opinion on this? I think it just clutters the viewport and makes constraint labels harder to place, especially with angles (fractional degree angles are so rare), but with distances as well.

jwesthues, 2016-04-15:
I think it's conventional to include trailing zeros on distance dimensions, since that roughly implies a tolerance. The number of digits after the decimal point is already configurable by the user globally. (And if you did want to drop trailing zeros, then why do it for integers only, and not e.g. 1.500 to 1.5?)

For angle, 45.00 degrees or whatever does get silly. The proper solution is probably to provide some kind of "significant figures" control per dimension (or per style, and assign styles to dimensions?), but dropping trailing zeros for angles seems like a reasonable interim step.

whitequark, 2016-04-15:
Good points.

whitequark, 2016-04-15:
Regarding styling, I'm slowly becoming less and less satisfied with the current style system. On the "it really has to be changed" side, things like hidden edges and outlines should be exported in however many millimeters ISO says, but rendered in pixels; printing pink constraints on my laser printer gets old pretty fast, too. So at the very least, there should be a way to specify a separate set of styles for exporting.

On the "this will make it nicer" side, styles have way too much irrelevant detail squashed into them and, similarly, promote unnecessary duplication. I would rather have the same base size for all text on the drawing and be able to change it in one place; conversely, many default styles have values that *seem* like they're used but they aren't, e.g. text size of the normals/workplane style is ignored and workplane names take the text parameters from the constraint style. Similarly with stipple etc.

Once nested sketches land the problem will become even more thorny because on one hand, clearly styles from the outer sketch should be inherited, but also clearly inner sketches should be able to override them.

I feel like this naturally yearns for some sort of "cascading stylesheet" structure, in the sense of a hierarchy of styles applied to a hierarchy of objects. UI is the hardest problem to tackle here but I have a few decent ideas about that.

Evil-Spirit, 2016-04-15:
@whitequark i think we could do SI prefixes, which is introduce logarithmic scale to precision and shorter label string:
```
1000 -> 1m
0.001 -> 1um

```

jwesthues, 2016-04-15:
Constraints do already have styles; it's just used only for text comments now. If you wanted to get fancy, then you could let the user configure significant figures (and unit display, and maybe even explicit tolerances, `1.23" ± 0.045"`) per style, and then make dimensions respect that. And then, now that the dimensions have tolerances, you could write code to do Monte Carlo sensitivity analyses and stuff, like by repeatedly regenerating the model at different randomly-chosen values within the tolerances...

It's indeed confusing that many values from predefined styles aren't used. The fix is presumably either to respect those values, or to suppress their display in the style screens as an ugly special case.

Some kind of inheritance might be nice, though potentially quite confusing to the user. Add an option to assign a parent style to each new style, and to set any parameter in the new style to either an explicit value or "inherit"?

Evil-Spirit, 2016-04-15:
@whitequark, yes, I also thinking about someting like CSS for SolveSpace's sytles.
We should also consider imported sketches with their styles; so the problem is complex;
We could refactor import to keep the real Sketch instance like resources(load once, able to use few times without pelanlty for loading) It will give ability to consider it's styles (checkbox for imported group "override styles/use original styles").

Evil-Spirit, 2016-04-15:
@jwesthues, what if we split into EntityStyle and DimStyle (like DXF).
There is no point to talk about TextSize inside entity styles,
And there is no point talk about precision, arrows and extension lines metric (to consider ANSI, DIN, GOST, ISO) inside entity styles.

jwesthues, 2016-04-15:
Separate entity and dimension styles might be nice. Since the current styles already apply to both, that may become an ugly change; it may be preferable to e.g. keep a single set of styles internally, but make it look like two separate lists in the UI.

Evil-Spirit, 2016-04-15:
@jwesthues, inheritance of the style would be good solution, but too complex.

whitequark, 2016-04-15:
> If you wanted to get fancy, then you could let the user configure significant figures (and unit display, and maybe even explicit tolerances, 1.23" ± 0.045") per style, and then make dimensions respect that.

Yep, I'd like this. I almost started implementing it just for display purposes, but eventually pushed to a later date.

> And then, now that the dimensions have tolerances, you could write code to do Monte Carlo sensitivity analyses

That would be quite useful indeed.

> Some kind of inheritance might be nice, though potentially quite confusing to the user. Add an option to assign a parent style to each new style, and to set any parameter in the new style to either an explicit value or "inherit"?

What I had in mind is inverting the relationship between style parameters and styles. The style would become just a label with a name and id; the style parameters could be assigned to one or more labels, e.g. just `solid-edge`, `solid-edge` + `display`, `solid-edge` + `exported`. The style parameters would be selected first in the order of increasing specificity (i.e. `solid-edge` + `display` wins over `solid-edge`) and then in the shallow-to-deep order in hierarchy (i.e. styles from an inner sketch win over styles from an outer sketch, or something).

The UI would be a list displaying the mapping between sets of labels and associated style parameters. Any parameters that aren't overridden for a particular set of labels will be omitted from display; and editing would be done with granularity. I.e. you could separately add the "line stroke" parameter group, the "text style" parameter group, the "tolerances" parameter group and so on, to avoid cluttering the editor.

When a set of labels has another set of labels that it inherits anything from, it would display those values inline as inherited, so that at all times one could simply look up the most specific set of labels for a particular display element (or be pointed to it by `Style Info` context menu command).

Evil-Spirit, 2016-04-15:
> it look like two separate lists in the UI
yes, just have Style::type (Style::DIMENSION_TYPE, Style::ENTITY_TYPE )

whitequark, 2016-04-15:
I think this solution I have is exactly equivalent to our current one (with some UI improvements such as hiding irrelevant parameter groups), is generic enough to handle many possible cases, exposes all information necessary to understand how it works directly, and is possible to implement efficiently (browsers had to figure out how to make lookups in stylesheets extremely fast, and we can just do the same thing).

jwesthues, 2016-04-15:
@whitequark, I like that, but also fear that any web designer would understand that easily, and no one else... Maybe we need some non-programmer mechanical designers to look at this.

whitequark, 2016-04-15:
@jwesthues OK. I will gather feedback when it's time to rework that.

jwesthues, 2016-04-15:
Honestly, I suspect that most users would rather just set cosmetic properties per entity or constraint, and not bother with styles at all, regardless of how much time that might save them in the end. (What fraction of users actually set up styles in word processing software, for example?)

Let users do that, by silently creating a new style behind the scenes, and then also let users assign entities to "same style as entity xxx"? That's easier at first, though it potentially gets confusing really fast.

Evil-Spirit, 2016-04-15:
Returning to the main theme:
We have to remove trailings by ourselves; there is no specific format.
http://stackoverflow.com/questions/15453470/print-float-double-without-trailing-zeros
To use %g we have to introuduce SI metric prefixes (to avoid scintific to appear).

Symbian9, 2016-04-15:
@whitequark
<blockquote>So at the very least, there should be a way to specify a separate set of styles for exporting</blockquote>
Please, try read again few my issuse, that posted before:
* https://github.com/whitequark/solvespace/issues/93
* https://github.com/whitequark/solvespace/issues/109

whitequark, 2016-04-15:
@Symbian9 Like I said in those issues, separate styles for export is a planned, necessary feature. But it will not be done before we have a proper separate drawing workbench.

whitequark, 2016-04-16:
Fixed in 1f16927f02.

closed 2016-04-16 by whitequark