whitequark/solvespace #114: Export to DXF convert circles to arcs/splines

Symbian9, 2016-04-15:
After discussing import DXF future, think we find export issue, where circles converted in arcs when exported to DXF.
@Evil-Spirit <a href="https://github.com/whitequark/solvespace/issues/106#issuecomment-210454953">wrote here</a>:
<blockquote>Circles just becomes an arcs.
This is an export issue, not the import.</blockquote>
I try find some info elsewhere about circles in DXF:
* http://www.autodesk.com/techpubs/autocad/acad2000/dxf/circle_dxf_06.htm
* http://moi3d.com/forum/lmessages.php?webtag=MOI&msg=3635.2
* https://pythonhosted.org/dxfwrite/entities/circle.html
* https://bugs.launchpad.net/inkscape/+bug/1489320
* http://forums.sketchup.com/t/creating-a-perfect-circle-to-cut-on-a-laser-cutter/1647

Evil-Spirit, 2016-04-15:
There is actually no problem to do it. Just need to separate export mode when we need to export 2d drawing, not the mode where we have to recognize bezier curves.

whitequark, 2016-04-15:
@Evil-Spirit I'm actually not sure how a separate export mode would help. It would still use a combination of sketches internally, wouldn't it? So, anything we can do for it, we can do now.

Symbian9, 2016-04-16:
@whitequark, maybe its possible check is circle orthogonal to 2D View before export:
* if true - export this as circle
* if false - export this as curve

whitequark, 2016-04-16:
@Symbian9  That's already supposed to happen. https://github.com/whitequark/solvespace/blob/for-upstream/src/exportvector.cpp#L342-L349

Symbian9, 2016-04-16:
@whitequark, so, what is going wrong in @Evil-Spirit test described in issue about import?

whitequark, 2016-04-16:
Dunno.

Evil-Spirit, 2016-04-16:
@whitequark 
When export now, we are operating on SBezier, where circles are split into arcs (4 arcs for circles).
To export them as true circles we must operate on Entities. so, we need export mode for exporting 2d basing on just current workplane entities (the pure 2d case).

Evil-Spirit, 2016-04-16:
@whitequark There is no point to go in direction of assembling arcs to circles; because for detail-views we using just segments, so we have not to export Entities, except the case where we want to export 2d workplane (without any meshes or edges)

whitequark, 2016-04-16:
Okay, but when does the code in https://github.com/whitequark/solvespace/blob/for-upstream/src/exportvector.cpp#L342-L349 activate? I'm confused. I thought it should work in this case.

whitequark, 2016-04-16:
Oh--that's actually what creates the arc. It creates four pi/2 arcs instead of one circle.

whitequark, 2016-04-16:
@Symbian9 Until I see a compelling use case where a circle is fine but four arcs are not, I'm not going to allocate any resources to this feature.

closed 2016-04-16 by whitequark