whitequark/solvespace #115: Chrash when right clicking on a face

pimdegroot, 2016-04-16:
Created a simple cube, selected one face, right clicked and solvespace crashed.

Error is:

Failed to look up item 00008003, searched 7 items
oops at line 302, file /home/pim/solvespace/src/dsc.h

Aborted (core dumped)

my linux build and the windows nightly of 16-04-2016 have the same problem.

Symbian9, 2016-04-16:
Yes, also give this bug now. Same with click on extruded lines.
- in GDB
<code>(gdb) run
Starting program: /usr/bin/solvespace 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeafad700 (LWP 4067)]
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
failed to look up item 00008003, searched 7 items
oops at line 302, file /home/travis/build/whitequark/solvespace/src/dsc.h

Program received signal SIGABRT, Aborted.
0x00007ffff3d9bf77 in raise () from /lib/x86_64-linux-gnu/libc.so.6</code>

Binary: https://github.com/whitequark/solvespace/releases/download/nightly.2016-04-16/solvespace_2.1_amd64.deb
OS: LinuxMint 16, x86_64

whitequark, 2016-04-17:
Fixed in b3df59576.

closed 2016-04-17 by whitequark