whitequark/solvespace #116: Flip normal when extruding & using difference

pimdegroot, 2016-04-16:
When using the extrude tool the default is to extrude outward with a union Boolean.

Switching to difference keeps the direction outward. What i would expect is that the direction flips inward so you can see that it makes a hole.

whitequark, 2016-04-17:
I definitely see the issue--this is an extra action every time one creates a hole, and on large sketches can be quite slow, too, since you cannot readily constrain it in the right orientation while the normal points in the wrong direction, and so you have to wait for the solver many times.

However, I see a problem with your solution: if there are any constraints on any entities created by the extrusion, and it's flipped over, then the sketch may become quite broken.

pimdegroot, 2016-04-17:
in that case make the default dimension negative when switching to difference, failing gracefully when there is a constraint preventing that.

whitequark, 2016-04-17:
> in that case make the default dimension negative when switching to difference

There's no "default dimension" for extrusions--when they are not constrained they are simply free to move along all degrees of freedom they have, like everything else.

@jwesthues Any idea how to do this cleanly?

jwesthues, 2016-04-17:
@whitequark Not really. You could flip the extrude vector only in special cases (totally unconstrained, one length dimension on line parallel to extrude direction, etc.), I guess.

whitequark, 2016-04-17:
Okay. I think what I'll do is implement flipping for the case where no constraints mention the entities from the current group, to address the most common case: extruding something you immediately know should be a hole.

Evil-Spirit, 2016-04-28:
@whitequark, can we just make one more button "create extruded hole" with "difference" by default and corresponding direction?

whitequark, 2016-04-28:
What's the problem with flipping normal?

Evil-Spirit, 2016-04-28:
@whitequark 
you just want to reverse extrusion every time when you switch between difference and other?

whitequark, 2016-04-28:
Yep

Evil-Spirit, 2016-04-28:
OK

whitequark, 2016-05-04:
@Evil-Spirit Any progress on this?

Evil-Spirit, 2016-05-04:
Not yet. I've just started after holydays, need to think. Can we just test number of constraints == 0 inside group before filpping?

whitequark, 2016-05-04:
OK, no problem.

I think that works, yes.

whitequark, 2016-05-06:
Fixed in 3b55303.

closed 2016-05-06 by whitequark