whitequark/solvespace #117: Enable workflow tool > select > confirm

pimdegroot, 2016-04-16:
The current way to use a tool in solvespace is to select the items you want to modify and then apply a tool.

In other cad programs there is a second workflow where you select a tool, select the items you want to apply it to and then confirm.

One example would be the dimension tool, which in the new workflow could work like this: Nothing selected > press D > select one or multiple lines or points > confirm with enter > have text floating under pointer > confirm location with click.

Extrude could work like this: nothing selected > extrude tool > select areas to extrude > confirm > confirm location with click.

whitequark, 2016-04-17:
So, SolveSpace can definitely be tedious at times when many constraints have to be created. Automatically creating point-coincident, point-on-line, horizontal and vertical constraints, which is currently done, helps quite a bit with that.

But I'm not sure how the workflows you suggest are supposed to work. For dimension constraints, this would create n dimensions, all with same value, right? Why not constrain them to the same length and then add one dimension constraint?

For extruding, right now extruding can only be done for a sketch-in-plane group, where the entire group is always extruded, regardless of selection, so I am also not sure how would that work.

pimdegroot, 2016-04-17:
the thing solvespace does now is to throw an error message at you when you have not selected the right amount of items, which breaks workflow.

The dimension tool would work like this: https://youtu.be/gu7Nm0ygaow?t=21m38s

in fusion 360 you start a sketch on a face with the outline of that face already drawn, so there is a need to select the right area, either inside or outside of your sketched items.

whitequark, 2016-04-17:
> The dimension tool would work like this: https://youtu.be/gu7Nm0ygaow?t=21m38s

Ah I see. Yes, this is definitely convenient when placing a large amount of dimensions.

> in fusion 360 you start a sketch on a face with the outline of that face already drawn, so there is a need to select the right area, either inside or outside of your sketched items.

There's a planned "Sketch on face" feature but it was postponed due to some internal issues. I'll look into seeding the newly created sketch this way.

Symbian9, 2016-04-17:
<blockquote>in fusion 360 you start a sketch on a face with the outline of that face already drawn, so there is a need to select the right area, either inside or outside of your sketched items.</blockquote>
Think, this future can be copyrighted, so be carefull with copying functionality.(I'm hope that I'm wrong in this case)

whitequark, 2016-04-17:
Ideas cannot be copyrighted, only code.

pimdegroot, 2016-04-17:
You can patent them though, especially in the us.

Evil-Spirit, 2016-04-17:
@pimdegroot, I am waiting when they patent the food-eating and will die. All who believe in patents.

whitequark, 2016-04-17:
> You can patent them though, especially in the us.

Well... yeah. But also FOSS has violated software patents since times immemorial. I'm willing to take the chance on this one, I guess.

whitequark, 2016-04-17:
@Evil-Spirit Strictly speaking, food eating is a prior art and therefore any patent on it would be invalid under the existing patent system.

leeoniya, 2016-04-17:
@whitequark You could probably patent a specific method of food eating. A particular chewing pattern perhaps...

Symbian9, 2016-04-17:
<blockquote>Ideas cannot be copyrighted, only code.</blockquote>
@whitequark, no, currently its POSSIBLE protect interaction methods too.
Read this: http://stackoverflow.com/questions/248606/are-uis-copyrighted

whitequark, 2016-04-17:
> The court ruled that, "Apple cannot get patent-like protection for the idea of a graphical user interface, or the idea of a desktop metaphor [under copyright law]..."

Symbian9, 2016-04-17:
@whitequark, also read this (from link above)
<blockquote>Read up on Apple vs. Microsoft. A UI is copyrightable as long as it is not the only way such an interface could possibly be created.</blockquote>

As I remember, "Drag&Drop" method really was copyrighted by Microsoft.

Evil-Spirit, 2016-04-30:
We can just implement "commands", which know their input types and number, then we can implement something like context-choosing of "commands" like pie menu. We can run selection one-by-one when we have no selection or immediately execute the command when we have enough properly-typed selected items.

whitequark, 2016-11-26:
Migrated to https://github.com/solvespace/solvespace/issues/121.

closed 2016-11-26 by whitequark