whitequark/solvespace #118: Export 2D View of sphere to PDF crash SolveSpace

Symbian9, 2016-04-17:
Latest build crashed when try export 2D view of sphere.
1) Start SolveSpace
2) Create circle (or closed arc as half of circle)
3) Lathe it to sphere
4) Try Export 2D View

RESULT: SolveSpace freeze and crash

whitequark, 2016-04-17:
@Evil-Spirit Stack overflow.
```
#0  0x00000000004c79f6 in SolveSpace::SBsp3::InsertConvex (this=0x88c66d0, meta=..., vertex=0x7ffffffe05f0, cnt=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:160
#1  0x00000000004c77f5 in SolveSpace::SBsp3::InsertConvexHow (this=0x88c6420, how=101, meta=..., vertex=0x7ffffffe05f0, n=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:142
#2  0x00000000004c7eba in SolveSpace::SBsp3::InsertConvex (this=0x88c6420, meta=..., vertex=0x7ffffffe05f0, cnt=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:200
#3  0x00000000004c77f5 in SolveSpace::SBsp3::InsertConvexHow (this=0x88c6270, how=101, meta=..., vertex=0x7ffffffe05f0, n=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:142
#4  0x00000000004c7eba in SolveSpace::SBsp3::InsertConvex (this=0x88c6270, meta=..., vertex=0x7ffffffe05f0, cnt=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:200
#5  0x00000000004c77f5 in SolveSpace::SBsp3::InsertConvexHow (this=0x88c60c0, how=101, meta=..., vertex=0x7ffffffe05f0, n=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:142
#6  0x00000000004c7eba in SolveSpace::SBsp3::InsertConvex (this=0x88c60c0, meta=..., vertex=0x7ffffffe05f0, cnt=4, instead=0x0)
    at /home/whitequark/Work/solvespace/src/bsp.cpp:200
...
```

Evil-Spirit, 2016-04-18:
@whitequark, stack depth exceed on the booleans using triangle mesh. This is issue of recursive algorithm, so we need to refactor it, but this is not so simple.

whitequark, 2016-04-18:
@Symbian9 For now, increase export chord tolerance.

Evil-Spirit, 2016-04-18:
@whitequark, what we can do now - just increase stack size in compilator settings.

whitequark, 2016-04-18:
That's very nonportable. There's a myriad of ways to do this under *nix.

whitequark, 2016-04-18:
The stack frames of `InsertConvex` are also excessively and unnecessarily large:
```
    bool isPos[MAX_VERTICES];
    bool isNeg[MAX_VERTICES];
    bool isOn[MAX_VERTICES];
    Vector vpos[MAX_VERTICES];
    Vector vneg[MAX_VERTICES];
```

Evil-Spirit, 2016-04-18:
Ok.  We can just make it smaller not changeing algorithm.

whitequark, 2016-04-18:
I'm not sure if that really helps--if a simple sphere is enough to trigger the misbehavior then what happens on a complex sketch?

Evil-Spirit, 2016-04-19:
@jwesthues There is logrithmic algorithm and I suppose what depth+1 twices the element number which can be handled, isn't it?

jwesthues, 2016-04-19:
The BSP algorithms are usually more or less logarithmic, but it's complicated to predict the exact depth since (a) triangles get split, depending whether they intersect the planes of other triangles, and (b) balancing spatial data structures is hard. A giant convex polyhedron (i.e., a polyhedron where every face's plane divides space between "that face, in the plane", and "all the other faces, on the same side") is a particularly bad special case. More complicated-looking sketches almost certainly perform much better.

The default stack size on Windows was insufficient, so my original build fixed that up to 8 MB (the `editbin` line). Other builds should presumably do something similar. Beyond that, yeah, you could (a) reduce the stack frame size with dynamic memory allocation, (b) try to balance the tree by choosing the next input triangle to insert intelligently, instead of just randomizing like now (but I don't think that would help much here, the convex thing), and/or (c) occasionally splitting against planes other than an input triangle, if none of the input triangles give a reasonably balanced partition (which would get you back to logarithmic performance for any input, including convex polyhedra; but you don't want to split against non-triangle planes more than necessary, since that also may split the triangles into more pieces, risking numerical problems).

jwesthues, 2016-04-19:
Oh, and while we're here: (a) both BSP trees and k-d trees are implemented, since the BSP trees "naturally" split the triangles in a way that makes Boolean operations on mesh or generation of a 2d view with correct z-order easy, and (b) the k-d trees do actually make an attempt to balance themselves, which could be copied and used in the BSP trees (since you can always partition against additional arbitrary planes in the BSP trees, at no cost other than some potential unnecessary splitting), though that should probably also be improved (e.g., by choosing the split plane offset `split[]` as the median with quickselect, not the mean as now).

Evil-Spirit, 2016-04-19:
@jwesthues, a long time ago I look into Quake BSP and implemented something the similar. As I remember, there is some balancing there (and probably better algorithm which is not cause stack overflow). May be you know other things we can look at?

jwesthues, 2016-04-19:
There is nothing you can do to balance the BSP of a convex region without inserting split planes other than the original faces. I don't know what Quake did. You can make the stack much bigger, and still consume a negligible amount of memory by modern standards, so I wouldn't worry much about that (though smaller would improve cache performance, of course).

whitequark, 2016-04-19:
OK. I will look into portably increasing stack size. There's `setrlimit(RLIMIT_STACK)`, which should work in theory...

jwesthues, 2016-04-19:
To be clear, your BSP tree in this unusual special case is perfectly unbalanced, a "BSP linked list". So the bigger stack should fix the crash, but improvement to the underlying algorithm would be valuable too...

whitequark, 2016-04-19:
@jwesthues The stack size in this case has to grow linearly with the number of triangles, right? E.g. if we increase it and the crashes stop, if someone uses two spheres, that's going to be a crash again. If so, looks to me that an improvement to the algorithm would be much more useful.

Evil-Spirit, 2016-04-19:
@whitequark, balancing will be great, but firstly we have to avoid recursion.

jwesthues, 2016-04-19:
@whitequark Tail-call optimization might be helping, but without that, yeah. You could rewrite the algorithm without recursion, but that just replaces the call stack with an explicit stack (which is potentially more compact, of course), and certainly doesn't help asymptotically. Balancing by introducing new split planes seems like the correct long-term answer.

Oh, and "two spheres" is no longer a convex region, so that probably performs better. "One sphere more finely meshed" doesn't, though.

whitequark, 2016-04-19:
> Tail-call optimization might be helping

I don't know what MSVC does, but on *nix the C calling convention is incompatible with TCO, so for any functions that are exported from the translation unit TCO is not done. (Probably for other reasons as well, e.g. that mutual recursion between `InsertConvex` and `InsertConvexHow`.)

jwesthues, 2016-04-19:
I'm not sure either, so I hedged with "might". An explicit stack (instead of recursion) would remove all that uncertainty, and potentially run a bit faster (but not asymptotically faster). The additional split planes would get you log(n) depth even with pathological input, probably making the former change unnecessary. That's more complex, though probably not too bad, maybe something to do in conjunction with the k-d tree refactoring discussed elsewhere.

Evil-Spirit, 2016-04-28:
>     bool isPos[MAX_VERTICES];
>     bool isNeg[MAX_VERTICES];
>     bool isOn[MAX_VERTICES];
>     Vector vpos[MAX_VERTICES];
>     Vector vneg[MAX_VERTICES];
This is not to blame, I have tested so much.
even so simple example like this:
```
void testStackDepth(int i = 0) {
    Vector v0 = Vector::From(0,0,0);
    Vector v1 = Vector::From(0,0,0);
    Vector v2 = v0.Plus(v1);
    testStackDepth(i + 1);
}
```
Can reach only ~4800 depth(only 4800 polygons for case of sphere). 

1. TCO will not shoot(or shoot depending on compiler and this is not for MSVC)
2. Also, I've tried to do something with goto (goto causes destructor call, but not stack free)
3. There is way to decompose into small functions, but this is nonsence considering the simple case.
4. There is only way - fully remaster algorithm

Evil-Spirit, 2016-04-28:
```
void	Build_BSP_Tree (BSP_tree *tree, list polygons)
{
   polygon   *root = polygons.Get_From_List ();
   tree->partition = root->Get_Plane ();
   tree->polygons.Add_To_List (root);
   list      front_list,
             back_list;
   polygon   *poly;
   while ((poly = polygons.Get_From_List ()) != 0)
   {
      int   result = tree->partition.Classify_Polygon (poly);
      switch (result)
      {
         case COINCIDENT:
            tree->polygons.Add_To_List (poly);
            break;
         case IN_BACK_OF:
            back_list.Add_To_List (poly);
            break;
         case IN_FRONT_OF:
            front_list.Add_To_List (poly);
            break;
         case SPANNING:
            polygon   *front_piece, *back_piece;
            Split_Polygon (poly, tree->partition, front_piece, back_piece);
            back_list.Add_To_List (back_piece);
            front_list.Add_To_List (front_piece);
            break;
      }
   }
   if ( ! front_list.Is_Empty_List ())
   {
      tree->front = new BSP_tree;
      Build_BSP_Tree (tree->front, front_list);
   }
   if ( ! back_list.Is_Empty_List ())
   {
      tree->back = new BSP_tree;
      Build_BSP_Tree (tree->back, back_list);
   }
}
```
ftp://ftp.sgi.com/other/bspfaq/faq/bspfaq.html#7.txt

Evil-Spirit, 2016-04-28:
@whitequark 
We can just avoid BSP tree building for DXF export (and in cases when faces are switched off). When exporting DXF we are not exporting any triangles, because to build SKd tree is enough for hidden line removing algorithm.

Evil-Spirit, 2016-04-28:
@whitequark ,
Please look at https://github.com/whitequark/solvespace/blob/master/src/export.cpp#L522.
Before it, we are building BSP, but if we don't want to export triangles, we just clear mesh.
If case of DXF, even if we are not clear the mesh, we do nothing with it. So, we can avoid this generation.

whitequark, 2016-04-28:
This issue is about PDF export

Evil-Spirit, 2016-04-28:
@whitequark, oops, sorry. but anyway, there is unlogical building of BSP which can be avoided to improve export performance.

Evil-Spirit, 2016-04-28:
What we have to do with THIS issue? Refactor the algorithm?

whitequark, 2016-04-28:
What about "balancing to introduce new split-planes"?

Evil-Spirit, 2016-04-28:
> What about "balancing to introduce new split-planes"?
Too complex for algorithm we don't want to support in the future. Just wasting of time.

Evil-Spirit, 2016-04-28:
> balancing to introduce new split-planes

@jwesthues,
There will be just the same as SKd tree(but with polygon splitting) instead of BSP?

jwesthues, 2016-04-28:
@Evil-Spirit, Correct, in the case of a convex polyhedron, the BSP tree with extra split planes would perform almost identically to the k-d tree, with the exception that any faces that intersect the extra split planes are split. Splitting along the plane of a face always partitions all the other faces on one side (since that's what "convex" means), so the tree depth becomes O(n) instead of ~O(log n), so that's not useful.

You could also reimplement the mesh Booleans and vector shaded 2d view generation to work without the BSP tree, like by putting the faces into a k-d tree, and splitting explicitly only when required. That's a lot of work to get right in all the special cases, though.

Evil-Spirit, 2016-04-30:
@jwesthues 
How do you think, can we get rid of InsertConvex?

1. It contains some code for arbitrary number of vertices, but used only for inserting quads
2. It contains duplicated code (like for triangels - classification, etc)

Instead of this:

(a) Can we just use inserting of two traiangles? I don't think it will be much slower, but much simplier.
We can just remove a big part of the code, simplify it, and avoid supporting it.
(b) Refactor it to work only with quads
(c) Remove InsertTriangle and make one universal function for triangles and quads (working with Vector[])

jwesthues, 2016-04-30:
`InsertConvex` only ever starts with quads, but it may deal with convex polygons with more vertices after the quads get split. It's not there by accident; splitting triangles vs. arbitrary polygons will generate a much worse mesh in some cases, increasing the risk of failure in subsequent operations. Do you see why?

jwesthues, 2016-04-30:
If not, try sketching a triangle in the plane. Then split it against a line. That gives you a smaller triangle and a quad. Triangulate the quad into two more triangles, as you propose.

Then, split the resulting three triangles against the line that intersects the most edges. Repeat this a few times, and you'll see how quickly the number of triangles (and probability that you get a really skinny one that breaks the tolerance tests) explodes. By splitting general convex polygons instead of triangles, we avoid this.

Evil-Spirit, 2016-04-30:
@jwesthues.

So, can we made universal function to start with triangle to avoid code duplicating for three and more vertices cases?

jwesthues, 2016-04-30:
Universal is probably okay. The output should be identical, and the speed penalty should be small.

Evil-Spirit, 2016-04-30:
Reworked in
https://github.com/Evil-Spirit/solvespace/commit/cff88d99a75e218a1af205d513586edcc5b7b762

Evil-Spirit, 2016-05-04:
I am working to make BspUtil smaller (now is 512 bytes), 30 min and i will finish, so wait until it.

Evil-Spirit, 2016-05-04:
done (512 -> 64 bytes) https://github.com/Evil-Spirit/solvespace/commit/c148805516951f14301e35194b71059802f19564

Evil-Spirit, 2016-05-17:
We have to beat the last one?

whitequark, 2016-05-17:
I'm just going to bump it to 3.0, since it requires so much work...

Evil-Spirit, 2016-05-17:
Actually I am ready to try while the beta is testing.

whitequark, 2016-05-17:
Rewriting BSP would kinda contradict the point of having a feature-frozen release candidate cycle.

I would *really* like to move on to refactoring, we already spent far more time making 2.1 than I wanted. I hoped to release it in late February...

whitequark, 2016-05-17:
There's a trivial geometric workaround, which is to make the shape concave, and most shapes are already convex anyway; as well as reducing the amount of triangles in concave shapes  And if it turns out that people would really like to see this in 2.1 then we can expand the stack, which couldn't possibly introduce new bugs.

Evil-Spirit, 2016-05-17:
OK, just go Stone-Scissor-Paper realease cycle. Stone was completed, refactoring will be scissor, and the bsp optimizing(and other performance/memory usage improvements) will be paper.

whitequark, 2016-05-17:
The... what... release cycle??

Evil-Spirit, 2016-05-17:
I don't know, call it what you will.

Symbian9, 2016-08-23:
Any news here?

Evil-Spirit, 2016-08-23:
@Symbian9, not yet.

whitequark, 2016-11-27:
@Evil-Spirit Can you rebase this on top of master?

Evil-Spirit, 2016-11-27:
@whitequark, ofc

Evil-Spirit, 2016-11-27:
rebased https://github.com/Evil-Spirit/solvespace/commits/master-fix-bsp

whitequark, 2016-11-27:
Merged.

closed 2016-11-27 by whitequark